snaptec/openWB #737: Einschalt-SoC für Sofortladen

tomac01, 2020-11-28:
Sachverhalt:
Das EVU wird initial im Sofortlade-Modus bis 80% geladen. Danach verbleibt das EVU an die openWB angeschlossen. Nach der zyklisch Abfrage des SoC vom EVU, die auf 5 Std. eingestellt ist, liegt der neue SoC bei 78%. Gem. Einstellung im Sofortlade-MMdus beginnt damit die Ladung des EVU bis wieder der SoC von 80% erreicht ist. Die Dauer der Ladung beträgt nur 1-2 Minuten. Mit erreichen des Maximal-SoC schaltet die Ladung wieder ab. 

Problem: sehr kurze Ladezyklen sind für die Fahrzeugbatterie nicht so gut und sollten vermieden werden.

Anforderung:
Als Nutzer möchte ich, dass die Sofortladung meines EVU optional erst bei Unterschreiten eines Minimal-SoCs erfolgt, wenn das EVU dauerhaft (-> ohne Trennung zwischen durch) an der openWB angeschlossen ist und der Maximal-SoC gem. zyklischer Ermittlung am EVU unterschritten wird. 

Erläuterung:
Dies verhindert, dass bei einer dauerhaften Verbindung mit der openWB im Sofortlade-Modus bei zyklischer Ermittlung des SoCs die Ladung auch bei nur geringem Unterschreiten des Maximal-SoC die Ladung startet.

Beispiele nach Umsetzung:
Ausgangslage: Die openWB ist auf Sofortladen mit Maximal-SoC 80%, Minimal-SoC für zyklische Ermittlung des SoC 70% und einer zyklischen Abfrage des SoC von 2 Std. konfiguriert.
1. Das EVU hat einen SoC von 75% und war von der openWB getrennt. Es wird nun angeschlossen. Die openWB initiiert die Ladung bis zum Max-SoC von 80%.
2. Das EVU ist nach einer Sofortladung wie unter 1. weiterhin an der openWB angeschlossen. Nach einer zyklischen Ermittlung des SoC beträgt dieser 78%. Die Ladung wird nicht gestartet, da der Minimal-SoC nicht erreicht wurde.
3. Situation wie unter 2. jedoch wird nach längerer Zeit bei der zyklischen Abfrage des EVU-SoC ein Wert von 69% ermittelt. Die openWB beginnt mit der Ladung, da der Minimal-SoC unterschritten wird. Die Ladung erfolgt bis zum erreichen des Maximal-SoC von 80%.

snaptec, 2020-12-01:
Ich kann die Argumentation nachvollziehen. Habe das Problem reell aber nie erlebt.
Selbst die hungrigen Teslas brauchen (sofern kein Sentry an ist) bis zu 24h um 1% zu verlieren. 
Um welches Fahrzeug geht es hier genau?

tomac01, 2020-12-02:
Es handelt sich um ein Tesla Model S 75 von 2019.

In der Anlage mal das Ladelog von den letzten Tagen:
[image]