snaptec/openWB #745: 11kW Einphasiges Laden mit 32A

atze09, 2020-11-30:
Hallo zusammen,
Kurzer Hintergrund, wir haben uns den Opel e-Corsa mit einphasigem Lader (bis 7,2A) und somit ohne die Option des 3-phasigem Laders.
Wir haben 10kWp PV auf dem Dach und würden zu 100% mit PV Strom laden, aufgrund unserer geringen KM-Leistung scheint, das auch nicht abwägig.

Nun zu Frage ist es möglich mit der 11Kw gedrosselten OpenWB, einphasig den Ladestrom bis 32A (7,2kW) zu regeln.
Wir werden die Wallbox mit 32A LS installieren um ggfs. später upzugraden.
Trifft wahrscheinlich nur wenige Nutzer, aber für mich wäre die Option echt der Bringer, somit könnten wir den PV Eigenverbrauch noch weiter erhöhen, ich werde mal einen feature request auf github stellen.

Das wäre ja eigentlich KfW konform, wenn diese auch 7,2kW einphasig könnten, obwohl nur 11kW.
Da die KfW ja nur einen 3-phasigen Anschluss vorschreibt, jedoch nicht einen maximalen Strom pro Phase.
Ansonsten könnte dies erst nach einem Jahr und Genehmigung vom Netzbetreiber geschehen, das wäre Schade.

snaptec, 2020-12-01:
> Nun zu Frage ist es möglich mit der 11Kw gedrosselten OpenWB, einphasig den Ladestrom bis 32A (7,2kW) zu regeln.

Leider nicht. Die 11kW Beschränkung heißt 16A je Phase.

> Das wäre ja eigentlich KfW konform, wenn diese auch 7,2kW einphasig könnten, obwohl nur 11kW.
Leider nicht. Die KfW fordert den dreiphasigen Anschluss, sprich 3x 16A = 11kW

> Da die KfW ja nur einen 3-phasigen Anschluss vorschreibt, jedoch nicht einen maximalen Strom pro Phase.
Die Formulierung liegt "genau" 11kW.
Natürlich können wir dies ändern. Ist dann nur nicht KfW-Konform.

atze09, 2020-12-01:
> Leider nicht. Die 11kW Beschränkung heißt 16A je Phase.

Und wo steht das? konnte diese Information nicht finden? Der 3-phasige Anschluss ist ja auch nur mit ,wo möglich, vorgeschrieben..Am besten mit Screenshot.

snaptec, 2020-12-01:
https://www.kfw.de/PDF/Download-Center/Förderprogramme-(Inlandsförderung)/PDF-Dokumente/6000004534_M_440_Ladestationen_Elektroautos.PDF

Teil 3. zweiter Punkt.
Sicherlich mag das jeder auslegen wie er möchte. Das obliegt nicht uns.

Wir liefern die openWB gern in 22kW Variante. Die Beschränkung kann auch der Elektriker bestätigen.

atze09, 2020-12-01:
> [https://www.kfw.de/PDF/Download-Center/Förderprogramme-(Inlandsförderung)/PDF-Dokumente/6000004534_M_440_Ladestationen_Elektroautos.PDF](https://www.kfw.de/PDF/Download-Center/F%C3%B6rderprogramme-(Inlandsf%C3%B6rderung)/PDF-Dokumente/6000004534_M_440_Ladestationen_Elektroautos.PDF)
> 
> Teil 3. zweiter Punkt.
> Sicherlich mag das jeder auslegen wie er möchte. Das obliegt nicht uns.
> 
> Wir liefern die openWB gern in 22kW Variante. Die Beschränkung kann auch der Elektriker bestätigen.

Danke für den Link, wie schon geschrieben, ist das Ganze sehr offen formuliert und in meinen Augen besteht hier nicht die Beschränkung auf 16A pro Phase.

Könnte Ich die Wallbox auch als 22kW Variante bestellen und selbst softwaretechnisch auf 11kW (3-phasig) und 7,2kW (1-phasig) beschränken und dann vom Elektriker bestätigen lassen?

Die zweite Möglichkeit ist halt, dass Ihr mir die 11KW Drosselung nach dem von der KfW vorgeschriebenem Jahr wieder aufhebt, hm.

atze09, 2020-12-02:
Nach erneutem Telefonat mit der KfW Technikabteilung gibt es keine Beschränkung des Ladestroms auf einer Phase.
Somit kann bis 11kW Summe über alle Phasen geladen werden wie man möchte, sofern die Schieflastgrenzen des EVU’s nicht verletzt werden, das muss der Elektriker prüfen und bestätigen.

Somit wäre mein Wunsch, dass die 11kW Variante einen einphasigen Ladestrom von 32A zulässt.
Wäre dies softwaretechnisch für mich persönlich anpassbar, oder muss das in der Programmierung geschehen, oder muss ich die 22kW Variante bestellen?
Vielen Dank schon einmal.

snaptec, 2020-12-04:
Hierfür muss die 22kW Variante bestellt werden. Die Software gibt es derzeit nicht her einphasig 32A zuzulassen, dreiphasig aber nicht.
Es ist nur möglich die Schieflastgrenze zu beachten, sprich einphasig max 20A, dreiphasig max 32A.

closed 2021-01-02 by snaptec