snaptec/openWB #761: Add Json parsing option for battery

tkoeberl, 2020-12-03:
It's possible for PV/EVU to use a json/jq, but not for the battery. Would be cool to have this.

DetMoerk, 2021-02-08:
How many different values must be extracted? Only one?

Best Regards 
Det

DetMoerk, 2021-02-08:
@benderl: Wenn ich das richtig sehe gibt es bei Batteriespeicher die Werte Speicherleistung und Speichersoc. 

Bedeutet 
1 URL und 2 Filter Bedingungen analog dem PV JSON Modul würden reichen oder?

Könnte ich mal angehen und in den Batterie Bereich einbauen.

VG
Det

benderl, 2021-02-09:
Das wäre super! Speicher benötigen nur die aktuelle Leistung und den SoC.
Leistung in Watt -> ramdisk/speicherleistung
SoC ans Integer -> ramdisk/speichersoc

DetMoerk, 2021-02-09:
@tkoeberl : I'm working on it. Can you send me a example json output from your battery?

Regards
Det

tkoeberl, 2021-02-09:
@DetMoerk : sure:
`{"Apparent_output":90,"BackupBuffer":"0","BatteryCharging":false,"BatteryDischarging":false,"Consumption_Avg":337,"Consumption_W":409,"Fac":50.016502380371094,"FlowConsumptionBattery":false,"FlowConsumptionGrid":true,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":false,"GridFeedIn_W":-333,"IsSystemInstalled":1,"OperatingMode":"2","Pac_total_W":-3,"Production_W":79,"RSOC":20,"RemainingCapacity_W":4528,"Sac1":90,"Sac2":null,"Sac3":null,"SystemStatus":"OnGrid","Timestamp":"2021-02-09 16:02:18","USOC":13,"Uac":228,"Ubat":186}`

DetMoerk, 2021-02-09:
As the Battery Module only have two values to hold, I guess 
The values Consumption_W = Speicherleistung 
and 
RSOC =Speichersoc
.

I'll try to set it up as a module.

Regards
Det

DetMoerk, 2021-02-09:
@tkoeberl: give it a test when the PR is merged.

# 1031

DetMoerk, 2021-02-09:
@tkoeberl: if you need to invert the value of e.g. Consumption_W, wite the filter: .Consumption_W***-1**

tkoeberl, 2021-02-10:
@DetMoerk : openwb still does not have a docker container installation?

DetMoerk, 2021-02-10:
I guess not. I'm not in the core developer Team, I only added some modules around, which I also need. 

Did you test the speicher_json module?

Regards
Det

benderl, 2021-02-10:
We have no plans to support a docker container right now. If you are familiar with docker and manage to set it up, you are welcome to share your modifications.

tkoeberl, 2021-02-10:
I'm really sorry. I already destroyed my openwb installation. Thats why I was asking for the docker container.

closed 2021-02-13 by benderl