snaptec/openWB #799: Awattar nimmt deutsche statt österreichische Preise

riesi77, 2020-12-16:
Aktiviert man die Awattar Einstellung und wählt "Österreich" als Land aus, so werden trotzdem die Preise aus Deutschland angezeigt.
Version 1.9.003

snaptec, 2020-12-16:
Genutzt wird die österreichische API. Die Angabe erfolgt netto. Daher ggf die Differenz?

riesi77, 2020-12-16:
[image]

riesi77, 2020-12-16:
[image]

riesi77, 2020-12-16:
[image]

riesi77, 2020-12-16:
Ja, das scheint zu stimmen. Erst war ich verwirrt, aber die Grafik nimmt ja einen unterschiedlichen Zeitbereich. Und die Nettoangabe auf der Homepage war mir auch neu.

closed 2020-12-17 by snaptec