snaptec/openWB #808: Usability Awattag

riesi77, 2020-12-21:
Ziel der Verwendung von Awattar sollte es sein, genaue Schwellen (0,x Cent) definieren zu können, um die günstigsten Stunden zum LAden nutzen zu können. Das ist derzeit nicht oder nur eingeschränkt möglich:

1. Preise auf der Homepage und in OpenWB unterscheiden sich von der Berechnungsart (Brutto/Netto). Besser wäre es, (ev. konfigurierbar), auch in OpenWB den Nettopreis einstellen zu können, so wie er unter https://www.awattar.com/services/charts/hourly angezeigt wird.
2. Die Eingabeform des Preises über Schieberegler - speziell am Handy - erlaubt es nur nach mehreren Versuchen, einen Wert 0,x Cent einzugeben. Besser wäre eine numerische Eingabemöglichkeit (numeric textbox)
3. Die Anzeige als Kurve ist nicht ideal. Es geht darum, geringe Unterschiede zwischen den Preisen herauszulesen, und das ist in dieser Form nur eingeschränkt möglich. Besser: Balkendiagramm mit numerischer Angabe des Werts. Oder eine tabellarische Auflistung.
4. Alternativ zur Eingabe eines Schwellwertes, wäre auch ein zeitgesteuertes Laden, allerdings nicht im Sinne von Zielladen (80% SOC bis ... Uhr), sondern z.B.  lade von 2:00 bis 3:30. Damit würden alle zeitabhängigen Tarife (z.B. Tibber) quasi generisch unterstützt werden

riesi77, 2021-01-08:
5) Wenn ich das richtig interpretiere, ergeben sich die OpenWB Preise durch den Awattar hourly Preis *1,19. Das ist für Österreich nicht richtig, da hier die Mehrwertsteuer 20% sind. Generell bin ich, wie unter 1) geschrieben, für eine Netto Auspreisung, damit eine 1:1 Vergleichbarkeit mit dem angezeigten Preis auf der Awattar Homepage gegeben ist.

benderl, 2021-01-20:
Hallo riesi77,
die Punkte 2, 3 sind bereits in der aktuelen Nightly angepasst worden.
Punkt 5 stelle ich gerade auch 20% um, da die komplette Einbindung überarbeitet wird.

Über eine Anzeige des Netto-Preises lässt sich auch diskutieren. Das wird eventuell später umgesetzt, wenn die allgemeine Anbindung zeitbasierter Stromtarife abgeschlossen ist. In diesem Zuge muss noch einiges im Hintergrund angepasst werden.

Gruß,
Lutz

riesi77, 2021-01-22:
Habe mir die aktuellen Änderungen in der Nightly angesehen. Kompliment! Toll finde ich, dass die Ladezeiträume, die sich aus der gewählten Preisschwelle ergeben, nun markiert werden. Herzlichen Dank!

benderl, 2021-01-22:
@MiOrt hat sich darum gekümmert.

closed 2021-01-22 by benderl

MiOrt, 2021-01-22:
Moin. Da ich momentan zusammen mit benderl ja das preisbasierte Laden etwas vorantreibe, vielleicht kurz ein Kommentar dazu. Ich habe schon einen PR gemacht, der +- Buttons einführt, damit der Preis einfacher eingestellt werden kann. Das ganze wird in den kommenden Tagen noch verfeinert und ausführlich getestet. In der nightly sollte das schon funktionieren. Das mit dem Preis Netto/Brutto ist im Blick, in wieweit das zeitnah berücksichtigt wird, kann ich noch nicht sagen. Die Tibber API liefert zB den Preis getrennt nach Börsenanteil und Steueranteil. Ich meine bei Awattar gesehen zu haben, dass der Anteil Steuern/Netzentgeld/etc. auch bei Vertragsschluss je nach Region unterschiedlich angezeigt wird. Vielleicht macht man da als Eingabefeld bei der Konfiguration, so dass der tatsächlich zu zahlende Preis dann im Frontend erscheint. Da wird sich demnächst sicher noch was tun.

riesi77, 2021-01-25:
Wie gesagt, die Implementierung der derzeitigen Nightly finde ich toll. Das sind richtig gute Verbesserung.
In Bezug auf das Ziel, einen sauberen Bruttopreis inklusive Netzkosten darzustellen, bin ich skeptisch. Die Netzkosten sind hier teilweise verbrauchsunabhängige Grundgebühren. Das ist also wirklich schwer darstellbar, ohne die anderen Verbraucher zu kennen. Da würde ich eher beim Nettopreis bleiben.