snaptec/openWB #836: Moduleconfiglp ist was faul

DetMoerk, 2021-01-02:
Hallo zusammen,

in der aktuellen Nightly ist die Konfig der Ladepunkte nicht mehr sauber möglich. Von mir ist da j der soc_eq Teil rein gekommen, aber es ist wohl auch an den JavaScript Funktionen geändert worden. 
Links Version aus dem aktuellen Snaptec Master Branch / rechts aus meinem Branch
[image]

Könnt ihr mal schauen, ob ich da was verbockt habe, oder ob es an der Änderung mit dem JS liegt?

VG
Det

benderl, 2021-01-02:
Die Funktionen sind in die helperfunctions.js gewandert. Der Aufruf hat sich etwas geändert und erwartet jetzt einen vollständigen JQ Selector. In diesem Fall einfach ein # am Anfang ergänzen für die ID der Divs.

Was funktioniert bei Dir nicht mehr? Ich kann hier keine Probleme feststellen.

benderl, 2021-01-02:
https://github.com/snaptec/openWB/pull/838

Da ist beim Auflösen des Merge Konflikts etwas schief gelaufen. Ist gleich behoben.

closed 2021-01-02 by benderl