snaptec/openWB #839: Mercedes SoC Modul - Link "HIER bei Mercedes Me anmelden" funktioniert nicht

StefanHesse, 2021-01-02:
Im heute gemergten Mercedes-Modul funktioniert der Link "HIER bei Mercedes Me anmelden" noch nicht. Man bekommt die Fehlermeldung:
"Oops
Looks like something is not right. Please contact your administrator.
400 - Invalid redirect_uri"
Ich bin mir ziemlich sicher, dass ich alle Werte korrekt eingetragen habe:
[image]
Da ich leider im openWB-Forum nichts mehr posten kann, probiere ich es jetzt hier. Sorry, wenn das der falsche Weg ist.

benderl, 2021-01-02:
@DetMoerk 
Kannst Du das bitte prüfen?

StefanHesse, 2021-01-02:
@DetMoerk 
Bei der Gelegenheit: In der Anleitung fehlt noch, wo man die "Fahrzeug ident" her bekommt. Ich habe sie jetzt aus meinem Mercedes Me Konto unter "Meine Fahrzeuge" --> "Fahrzeug verwalten" --> "Fahrzeugdaten" entnommen und hoffe, dass das die richtige ist.

DetMoerk, 2021-01-02:
Ich prüfe das nachher.

VG
Det

DetMoerk, 2021-01-02:
@StefanHesse: wird deine openwb in deinem Netzwerk sauber unter dem Namen openwb aufgelöst?

was sagt denn ein nslookup openwb auf deinem Rechner mit dem du die Einrichtung machst?
kommt da die IP Adresse deiner openWB Box zurück?

VG
Det

DetMoerk, 2021-01-02:
@StefanHesse : FIN (VIN) sollte passen. In meiner Mercedes Me App ist das unter Garage / Fahrzeuginformationen und beginnt auch mit W und endet mit einer Zahl.

Der Eintrag in der Mercedes Developer Site unter der API für Callback URL muss identisch mit dem Eintrag im Modul sein! Die Fehlermeldung deutet eigentlich darauf hin, dass der Eintrag zwischen openWB und der Mercedes App nicht identisch sind.

VG
Det

DetMoerk, 2021-01-02:
@StefanHesse: evtl. falsches Protokoll http:// <--> https:// ?

DetMoerk, 2021-01-02:
@benderl: Ichhabe noch ein Thema gefunden. Die Dateien zur Persisitierung (acc_tok_lp1, acc_tok_lp2, ref_tok_lp1, ref_tok_lp2, expires_lp1, expires_lp2) werden mit der user pi, Berechtigung 744 im Filesystem angelegt. Wenn der Apache nicht als user pi läuft macht das Probleme. 

@StefanHesse : Wenn eine komplett leere Seite zurück kommt, dann läufst du in das Berechtigungsproblem. Gib mir Bitte Bescheid, wenn du die Callback URLs überprüft hast. Eigentlich sollte eine Seite zurück kommen mit 
[image]

Ich bekomme meine echte openWB m Montag. Ich versuche die mal im Schnellgang nur mit einer Phase in Betrieb zu nehmen und schaue mir das dann mal dort an.

StefanHesse, 2021-01-02:
@DetMoerk: Hier die Antworten:
**wird deine openwb in deinem Netzwerk sauber unter dem Namen openwb aufgelöst?**
Ja, ich habe 2 Boxen. Die 2. Box habe ich in der FritzBox auf "openwb2" umbenannt. Allerdings stelle ich gerade fest, dass bei einem "nslookup openwb" von der FritzBox nicht nur die IPv6-Adresse, sondern auch 2 IPv4-Adressen zurück geliefert werden. Die 2. IPv4-Adresse ist die Adresse der 2. Box. Leider kann man die openWB ja nicht wirklich umbenennen, daher habe ich den Kunstgriff mit der FritzBox gemacht und bisher hatte ich auch nie irgendwelche Probleme. Ich probiere das aber gern mal mit der IP-Adresse anstelle des Namens, um das auszuschließen.

**Der Eintrag in der Mercedes Developer Site unter der API für Callback URL muss identisch mit dem Eintrag im Modul sein!**
Ich habe in der Developer Site 2 URLs eingetragen:
http://openwb/openWB/modules/soc_eq/callback_lp1.php
http://openwb/openWB/modules/soc_eq/callback_lp2.php
Da ich ja 2 Ladepunkte habe, dachte ich ich sollte da auch beide eintragen. Ich kann das aber auch mal nur mit dem 2. probieren, an den ich aktuell mein Auto angeschlossen habe.

**evtl. falsches Protokoll http:// <--> https:// ?**
Unter https sind meine openWB gar nicht zu erreichen, daher habe ich überall http genommen.

**Wenn eine komplett leere Seite zurück kommt, dann läufst du in das Berechtigungsproblem.**
Die Seite ist nicht leer, es sieht so aus:
[image]

Ich probiere es jetzt mal nur mit dem LP2 und mit den IP-Adressen. Gib mir ein paar Minuten :-)

StefanHesse, 2021-01-02:
So, habe jetzt wie oben geschrieben auf IP-Adresse umgestellt und nur die URL http://192.168.178.43/openWB/modules/soc_eq/callback_lp2.php auf der Developer Seite und in der openWB eingetragen.
Jetzt bekomme ich eine komplett leere Seite und nicht mehr die Fehlermeldung.

DetMoerk, 2021-01-02:
Komplett leer ist leider auch nicht gut. Jetzt bist du im Berechtigungsproblem, da leider der Apache wohl doch nicht unter dem gleichen Benutzer läuft wie der Rest der openWB. Was in einer normalen Raspi Installation normal ist. Ich schau mal ob ich das im Modul einbauen kann, dass die Berechtigungen der Dateien angepasst werden. Du hast nicht zufällig eine openWB mit SSH Zugriff?

VG
Det

StefanHesse, 2021-01-02:
Leider nein, ich habe keinen root-Zugriff auf die Boxen. Sonst hätte ich meine 2. Box auch schon längest auf "openwb2" umbenannt, um nicht 2 Geräte mit gleichem Namen im Netzwerk zu haben.
Meine Bitte die Box auch in der Weboberfläche ohne Rootzugriff umbenennen zu können ist bei openWB hoffentlich noch "in Arbeit" :-)

DetMoerk, 2021-01-02:
Sorry, ich glaube dann muss ich dich noch vertrösten. Habe wie gesagt aktuell nur ne Installation auf nem Raspi und ich bin schon in das Thema gelaufen aber meine Nachfrage in Richtung Benutzer und Berechtigungen ist entweder unter gegangen oder nicht ganz durchdrungen worden. Aber das bekomme ich noch hin.

VG
Det

benderl, 2021-01-02:
@DetMoerk 
Der Apache läuft nie als user pi. Die Prozesse der openWB laufen teilweise als pi, teilweise root. Mit Version 2.x wird das noch geradegezogen. Stand jetzt müssen wir damit leben und die Dateirechte entsprechend großzügiger setzen.

@StefanHesse 
Das ist auf der Todo-Liste für kommende Woche.

DetMoerk, 2021-01-02:
@benderl : Deshalb hatte ich bei dem Beispiel soc_modul danach gefragt, weshalb da ein sudo vor dem python Aufruf ist und ob alle Prozesse mit dem Benutzer Pi laufen. 
Könnt ihr beim Deployen auf die Boxen die Berechtigungen der Dateien acc_tok_lp1, acc_tok_lp2, ref_tok_lp1, ref_tok_lp1, expires_lp1, expires_lp2 im soc_eq Verzeichnis  auf 646 oder 666 setzen? 

VG
Det

DetMoerk, 2021-01-04:
@benderl: Noch eine Frage zur Ausgabe von Problemen mit einem SoC Modul. Wei könnten die evtl. auftretenden Fehler (falsche Credentials, Falsche FIN ...) in der Wallbox dargestellt werden, damit der Kunde das mit bekommt?
Ich habe in der Zwischenzeit noch einiges erweitert und werte auch die Fehlercodes aus.
VG
Det

benderl, 2021-01-04:
Das wird bitte in das soc.log geschrieben.
Aktuell gibt es sonst keine Möglichkeit der Rückmeldung. Ich werde da aber etwas einbauen, wenn ich mir das Modul ansehe, vermutlich aber frühestens morgen.

DetMoerk, 2021-01-04:
In das soc.log schreibe ich das jetzt auch rein. Warte bitte, ich bin gerade am Ändern des Moduls, habe auch entsprechende Log Meldungen eingebaut. Aus dem soc.log kann man dann den entsprechenden Fehlerstatus auch raus lesen. 
Gibt es einen Wert für SoC, wenn die Abfrage schief geht? Also soll SoC auf 0 oder auf 100 gesetzt werden, wenn der Wert nicht ermittelt werden kann, damit beim Laden nichts schief geht? 

VG
Det

DetMoerk, 2021-01-04:
@: hast du einen Tipp, wie man das Python Format für den Timestamp so einstellt, dass er identisch mit dem bash Timestamp Format des main.sh ist? Das RFC3339 ist unter python leider auf Millisekunden ausgerichtet.

VG
Det

benderl, 2021-01-04:
@DetMoerk schau mal in der mqttsub.py nach: https://github.com/snaptec/openWB/blob/b4112b737f2c050fba42eea4cd6a46c082eea96d/runs/mqttsub.py#L85
Da fehlt zwar die Zeitzone, sollte als Basis aber reichen.

benderl, 2021-01-04:
`%z` ist das Offset: https://www.programiz.com/python-programming/datetime/strftime

benderl, 2021-01-04:
> In das soc.log schreibe ich das jetzt auch rein. Warte bitte, ich bin gerade am Ändern des Moduls, habe auch entsprechende Log Meldungen eingebaut. Aus dem soc.log kann man dann den entsprechenden [...]

Das geht aktuell munter drunter und drüber. Entweder wird kein Update gemacht oder auf 0% gesetzt. Ich bin für die erste Variante in Verbindung mit einer Kennzeichnung und Logeinträgen, dass etwas schief gelaufen ist.

DetMoerk, 2021-01-04:
> 
> 
> `%z` ist das Offset: https://www.programiz.com/python-programming/datetime/strftime

Leider ist in dem Fall die Zeitzone als +0100, in der Bash ist die Zeitzone +01:00. 

Das sind gerade Einträge aus meinem soc.log

2021-01-04 14:41:26+01:00:                                 <-- bash timestamp=`date --rfc-3339=seconds`

2021-01-04T14:41:24.516239+01:00:                    <-- python3 local_time.isoformat() 
2021-01-04 14:41:24+0100:                                  <-- python3 local_time.strftime(format = "%Y-%m-%d %H:%M:%S%z")

Passt leider alles nicht so richtig zusammen. 

VG
Det

benderl, 2021-01-04:
Dann nimm doch das komplette Isoformat. Ich passe das in den anderen Skripten an. Ist vielleicht der einfachste Weg.

DetMoerk, 2021-01-04:
Ich habe jetzt local_time.strftime(format = "%Y-%m-%d %H:%M:%S%z") eingebaut und den fehlenden : durch ein Lerzeichen dahinter ausgeglichen. Damit passt es einigermaßen zusammen. 

2021-01-04 15:50:04+01:00: Lp2: Nothing to do yet. Incrementing timer. 3 < 5
2021-01-04 15:50:06+01:00: Lp1: Nothing to do yet. Incrementing timer. 2 < 5
2021-01-04 15:50:14+01:00: Lp2: Nothing to do yet. Incrementing timer. 4 < 5
2021-01-04 15:50:16+01:00: Lp1: Nothing to do yet. Incrementing timer. 3 < 5
2021-01-04 15:50:23+01:00: Lp2: Requesting SoC
2021-01-04 15:50:25+01:00: Lp1: Nothing to do yet. Incrementing timer. 4 < 5
2021-01-04 15:50:28+**0100 :** Lp2: SOC: 52 RANGE: 24
2021-01-04 15:50:34+01:00: Lp2: Nothing to do yet. Incrementing timer. 0 < 5
2021-01-04 15:50:36+01:00: Lp1: Requesting SoC
2021-01-04 15:50:39+**0100 :** Lp1: SOC: 52 RANGE: 24
2021-01-04 15:50:44+01:00: Lp2: Nothing to do yet. Incrementing timer. 1 < 5
2021-01-04 15:50:46+01:00: Lp1: Nothing to do yet. Incrementing timer. 0 < 5
2021-01-04 15:50:54+01:00: Lp2: Nothing to do yet. Incrementing timer. 2 < 5

VG
Det

DetMoerk, 2021-01-04:
@benderl Kannst du mir noch meine Frage beantworten:
Könnt ihr beim Deployen auf die Boxen die Berechtigungen der Dateien acc_tok_lp1, acc_tok_lp2, ref_tok_lp1, ref_tok_lp1, expires_lp1, expires_lp2 im soc_eq Verzeichnis auf 646 oder 666 setzen?

-rw-r--rw- 1 pi pi   42 Jan  4 14:17 ref_tok_lp2
-rw-r--rw- 1 pi pi   10 Jan  4 14:17 expires_lp2
-rw-r--rw- 1 pi pi 1178 Jan  4 14:17 acc_tok_lp2
-rw-r--rw- 1 pi pi   42 Jan  4 14:32 ref_tok_lp1
-rw-r--rw- 1 pi pi   10 Jan  4 14:32 expires_lp1
-rw-r--rw- 1 pi pi 1178 Jan  4 14:32 acc_tok_lp1

Wenn ich wüsste, wem die Dateien initial gehören, könnte ich das auch in den Modulen entsprechend einbauen, dass die Berechtigungen entsrpechend gesetzt werden. Aber meine Box hätte heute kommen sollen ist aber noch nicht da :-(.

VG
Det

DetMoerk, 2021-01-04:
@StefanHesse : Ich habe die Ermittlung der FIN in die Doku hier im Wiki eingebaut. Passt das für dich?

VG
Det

benderl, 2021-01-04:
> @benderl Kannst du mir noch meine Frage beantworten:
> Könnt ihr beim Deployen auf die Boxen die Berechtigungen der Dateien acc_tok_lp1, acc_tok_lp2, ref_tok_lp1, ref_tok_lp1, expires_lp1, expires_lp2 [...]

Habe ich hier lokal schon angepasst. Kommt direkt nach Deinem PR rein.

DetMoerk, 2021-01-04:
#844

StefanHesse, 2021-01-04:
@DetMoerk 
Sieht ganz gut aus, ich habe nur 2 Dinge gefunden:
1. Am Ende des Abschnitts "Find your FIN" fehlt ein ")".
2. In der Android-App gibt es keinen Menüpunkt "Garage". Wie es auf dem iPhone aussieht kann ich nicht sagen, ich vermute aber mal Du hast ein iPhone und da ist es evtl. etwas anders. In der Android-App muss man im Menü auf sein Auto klicken und kommt dann zur Garage, die dann auch wieder so heißt, aber ich habe zuerst verzweifelt einen Menüpunkt gesucht, der so heißt.

DetMoerk, 2021-01-04:
Hmm, habe auch ein Android. Ja, stimmt ist ein bisschen blöd. Ich schau mal.

VG
Det

StefanHesse, 2021-01-04:
Schreib doch einfach in Klammern dahinter "(erreichbar über das Menü und dann einen Klick auf das entsprechende Fahrzeug)"
Dann sollte es klar sein. Ich frage gern meine Tochter noch mal danach, ob es auf dem iPhone genauso aussieht.

StefanHesse, 2021-01-04:
So, habe mir das gerade über Skype Bildschirmteilen auf dem iPhone meiner Tochter angeschaut.
Auf einem iPhone sieht die App exakt so aus wie auf meinem Android. Die Beschreibung kann also auch für die iPhone-App genommen werden.

StefanHesse, 2021-01-05:
@DetMoerk , @benderl 
Das Problem ist vermutlich noch nicht gelöst. Ich habe jetzt ein Update gemacht und alles noch mal eingerichtet. Bei Callback habe ich folgendes eingetragen:
http://192.168.178.43/openWB/modules/soc_eq/callback_lp2.php

Die IP 192.168.178.43 ist meine Hauptbox, der Ladepunkt 2 ist aber der an der externen Box. Ich hoffe, dass das so richtig ist.
Wenn ich jetzt auf "HIER bei Mercedes Me anmelden" klicke, dann öffnet sich das Fenster, wo ich erlauben kann, dass diese App Zugriff bekommt. Das habe ich getan und dann kommt wie in der Beschreibung beschrieben ein leeres Fenster, welches ich dann geschlossen habe.

In der Übersicht sehe ich dann einen SoC-Wert, der aber immer bei 0% verbleibt. Mein Auto ist aktuell zu 100% geladen.
Mache ich was falsch oder haben wir noch einen anderen Bug?

DetMoerk, 2021-01-05:
@StefanHesse : Kannst du mal http://192.168.178.43/openWB/ramdisk/soc.log aufrufen und ein paar Zeilen hier rein stellen?
Bitte FIN raus lassen, falls das drin wäre.

VG
Det

DetMoerk, 2021-01-05:
Leeres Fenster dürfte nur kommen, wenn die Berechtigungen auf den Dateien für die Token nicht korrekt sind. Hatte gehofft, dass wir das auch gefixt bekommen,

VG
Det

StefanHesse, 2021-01-05:
@DetMoerk 
Vorher habe ich ja nur ein leeres Fenster bekommen. Nach dem heutigen Update kam ja erstmalig das Anmeldefenster von Mercedes. Erst nach dem dortigen Anmelden und klicken auf OK wurde ich dann auf ein neues leeres Fenster weitergeleitet. Jetzt kann ich erneut auf den Link klicken und es kommt sofort ein leeres Fenster so wie vor dem Update. Es scheint als ginge der Link nur einmal.
Hier das Log mit anonymisierter FIN:

2021-01-02 09:40:33+01:00: Lp2: Requesting SoC
SOC URL: https://api.mercedes-benz.com/vehicledata/v2/vehicles/VIN/containers/electricvehicle
2021-01-02 09:40:43+01:00: Lp2: Nothing to do yet. Incrementing timer. 0 < 360
2021-01-02 09:40:53+01:00: Lp2: Nothing to do yet. Incrementing timer. 1 < 360
2021-01-02 09:41:02+01:00: Lp2: Nothing to do yet. Incrementing timer. 2 < 360
2021-01-02 09:41:12+01:00: Lp2: Nothing to do yet. Incrementing timer. 3 < 360
2021-01-02 09:41:22+01:00: Lp2: Nothing to do yet. Incrementing timer. 4 < 360
2021-01-02 09:41:32+01:00: Lp2: Nothing to do yet. Incrementing timer. 5 < 360
2021-01-02 09:41:42+01:00: Lp2: Nothing to do yet. Incrementing timer. 6 < 360
2021-01-02 09:41:52+01:00: Lp2: Nothing to do yet. Incrementing timer. 7 < 360
2021-01-02 09:42:02+01:00: Lp2: Nothing to do yet. Incrementing timer. 8 < 360
2021-01-02 09:42:12+01:00: Lp2: Nothing to do yet. Incrementing timer. 9 < 360
2021-01-02 09:42:23+01:00: Lp2: Nothing to do yet. Incrementing timer. 10 < 360
2021-01-02 09:42:32+01:00: Lp2: Nothing to do yet. Incrementing timer. 11 < 360
2021-01-02 09:42:42+01:00: Lp2: Nothing to do yet. Incrementing timer. 12 < 360
2021-01-02 09:42:52+01:00: Lp2: Nothing to do yet. Incrementing timer. 13 < 360
2021-01-02 09:43:03+01:00: Lp2: Nothing to do yet. Incrementing timer. 14 < 360
2021-01-02 13:11:43+01:00: Lp2: Requesting SoC
SOC URL: https://api.mercedes-benz.com/vehicledata/v2/vehicles/VIN/containers/electricvehicle
2021-01-02 13:11:53+01:00: Lp2: Nothing to do yet. Incrementing timer. 0 < 360
2021-01-02 13:12:03+01:00: Lp2: Nothing to do yet. Incrementing timer. 1 < 360
2021-01-02 13:12:13+01:00: Lp2: Nothing to do yet. Incrementing timer. 2 < 360
2021-01-02 13:12:23+01:00: Lp2: Nothing to do yet. Incrementing timer. 3 < 360
2021-01-02 13:12:33+01:00: Lp2: Nothing to do yet. Incrementing timer. 4 < 360
2021-01-02 13:12:43+01:00: Lp2: Nothing to do yet. Incrementing timer. 5 < 360
2021-01-02 13:12:53+01:00: Lp2: Nothing to do yet. Incrementing timer. 6 < 360
2021-01-02 13:13:02+01:00: Lp2: Nothing to do yet. Incrementing timer. 7 < 360
2021-01-02 13:13:12+01:00: Lp2: Nothing to do yet. Incrementing timer. 8 < 360
2021-01-02 13:13:22+01:00: Lp2: Nothing to do yet. Incrementing timer. 9 < 360
2021-01-02 13:13:32+01:00: Lp2: Nothing to do yet. Incrementing timer. 10 < 360
2021-01-02 13:13:42+01:00: Lp2: Nothing to do yet. Incrementing timer. 11 < 360
2021-01-02 13:13:52+01:00: Lp2: Nothing to do yet. Incrementing timer. 12 < 360
2021-01-02 13:14:02+01:00: Lp2: Nothing to do yet. Incrementing timer. 13 < 360
2021-01-02 13:14:12+01:00: Lp2: Nothing to do yet. Incrementing timer. 14 < 360
2021-01-02 13:14:23+01:00: Lp2: Nothing to do yet. Incrementing timer. 15 < 360
2021-01-02 13:14:32+01:00: Lp2: Nothing to do yet. Incrementing timer. 16 < 360
2021-01-02 13:14:42+01:00: Lp2: Nothing to do yet. Incrementing timer. 17 < 360
2021-01-02 13:14:52+01:00: Lp2: Nothing to do yet. Incrementing timer. 18 < 360
2021-01-02 13:15:03+01:00: Lp2: Nothing to do yet. Incrementing timer. 19 < 360
2021-01-02 13:15:13+01:00: Lp2: Nothing to do yet. Incrementing timer. 20 < 360
2021-01-02 13:15:23+01:00: Lp2: Nothing to do yet. Incrementing timer. 21 < 360
2021-01-02 13:15:33+01:00: Lp2: Nothing to do yet. Incrementing timer. 22 < 360
2021-01-02 13:15:43+01:00: Lp2: Nothing to do yet. Incrementing timer. 23 < 360
2021-01-02 13:15:53+01:00: Lp2: Nothing to do yet. Incrementing timer. 24 < 360
2021-01-02 13:16:03+01:00: Lp2: Nothing to do yet. Incrementing timer. 25 < 360
2021-01-02 13:16:13+01:00: Lp2: Nothing to do yet. Incrementing timer. 26 < 360
2021-01-02 13:16:23+01:00: Lp2: Nothing to do yet. Incrementing timer. 27 < 360
2021-01-02 13:16:33+01:00: Lp2: Nothing to do yet. Incrementing timer. 28 < 360
2021-01-02 13:16:43+01:00: Lp2: Nothing to do yet. Incrementing timer. 29 < 360
2021-01-02 13:16:53+01:00: Lp2: Nothing to do yet. Incrementing timer. 30 < 360
2021-01-02 13:17:02+01:00: Lp2: Nothing to do yet. Incrementing timer. 31 < 360
2021-01-02 13:17:12+01:00: Lp2: Nothing to do yet. Incrementing timer. 32 < 360
2021-01-02 13:17:22+01:00: Lp2: Nothing to do yet. Incrementing timer. 33 < 360
2021-01-02 13:17:32+01:00: Lp2: Nothing to do yet. Incrementing timer. 34 < 360
2021-01-02 13:17:42+01:00: Lp2: Nothing to do yet. Incrementing timer. 35 < 360
2021-01-02 13:17:52+01:00: Lp2: Nothing to do yet. Incrementing timer. 36 < 360
2021-01-02 13:18:02+01:00: Lp2: Nothing to do yet. Incrementing timer. 37 < 360
2021-01-02 13:18:13+01:00: Lp2: Nothing to do yet. Incrementing timer. 38 < 360
2021-01-02 13:18:23+01:00: Lp2: Nothing to do yet. Incrementing timer. 39 < 360
2021-01-02 13:18:33+01:00: Lp2: Nothing to do yet. Incrementing timer. 40 < 360
2021-01-02 13:18:42+01:00: Lp2: Nothing to do yet. Incrementing timer. 41 < 360
2021-01-02 13:18:53+01:00: Lp2: Nothing to do yet. Incrementing timer. 42 < 360
2021-01-02 13:19:03+01:00: Lp2: Nothing to do yet. Incrementing timer. 43 < 360
2021-01-02 13:19:13+01:00: Lp2: Nothing to do yet. Incrementing timer. 44 < 360
[... cut ...]
2021-01-05 11:24:43: Lp2: Nothing to do yet. Incrementing timer. 60 < 360
2021-01-05 11:24:53: Lp2: Nothing to do yet. Incrementing timer. 61 < 360
2021-01-05 11:25:03: Lp2: Nothing to do yet. Incrementing timer. 62 < 360
2021-01-05 11:25:13: Lp2: Nothing to do yet. Incrementing timer. 63 < 360
2021-01-05 11:25:23: Lp2: Nothing to do yet. Incrementing timer. 64 < 360
2021-01-05 11:25:33: Lp2: Nothing to do yet. Incrementing timer. 65 < 360
2021-01-05 11:25:43: Lp2: Nothing to do yet. Incrementing timer. 66 < 360

DetMoerk, 2021-01-05:
Dein Browser speichert die Anmeldedaten bei Mercedes Me. Daher kommt nur beim ersten Aufruf die Anmeldemaske, danach nicht mehr. Das leere Fenster macht mich noch stutzig. Aber das Log sieht nicht soo schlecht aus. Er hat zumindest die Meldung "Acc Token Expired" und dann keine Fehlermeldung danach. Sieht für mich so aus, als ob die Token jetzt sauber in den Dateien landen. 
Return Code 204 ist blöd ( SoC Request Fehlgeschlagen unbekannter Code: 204), den habe ich nicht abgefangen. 20x Bereich sind gültige Rückmeldungen. Bedeutet also, dass alles korrekt eingerichtet ist, der Zugriff auch funktioniert. Ich bräuchte die JSON Rückgabe für den 204 Wert. 

VG
Det

DetMoerk, 2021-01-05:
In der API steht das folgende zu RC 204.

If no data is available for the resource, then the endpoint returns http status code 204

Kannst du das Fahrzeug ma starten und dann schauen, ob du einen SoC Wert bekommst?

VG
Det

StefanHesse, 2021-01-05:
@DetMoerk 
Ich habe das Auto mal gestartet und bin sogar mal ne Runde um den Block gefahren, bekomme aber leider bis jetzt noch keinen Wert. Wenn ich auf den Aktualisieren-Knopf im Webinterface drücke, dann dreht sich das Ding ohne Ende, aber es bleibt bei 0%.

DetMoerk, 2021-01-05:
Bitte noch mal den letzten Ausschnitt aus dem Log reinstellen. Welches Auto hast du? Baujahr?

VG
Det

StefanHesse, 2021-01-05:
@DetMoerk 
Ich habe einen Mercedes GLC 300e 4MATIC, aus dem Juli 2020.
Hier die an das obige Log anschließenden Zeilen:

2021-01-05 11:25:53: Lp2: Nothing to do yet. Incrementing timer. 67 < 360
2021-01-05 11:26:03: Lp2: Nothing to do yet. Incrementing timer. 68 < 360
2021-01-05 11:26:13: Lp2: Nothing to do yet. Incrementing timer. 69 < 360
2021-01-05 11:26:23: Lp2: Nothing to do yet. Incrementing timer. 70 < 360
2021-01-05 11:26:33: Lp2: Nothing to do yet. Incrementing timer. 71 < 360
2021-01-05 11:26:43: Lp2: Nothing to do yet. Incrementing timer. 72 < 360
2021-01-05 11:26:53: Lp2: Nothing to do yet. Incrementing timer. 73 < 360
2021-01-05 11:27:03: Lp2: Nothing to do yet. Incrementing timer. 74 < 360
2021-01-05 11:27:13: Lp2: Nothing to do yet. Incrementing timer. 75 < 360
2021-01-05 11:27:23: Lp2: Nothing to do yet. Incrementing timer. 76 < 360
2021-01-05 11:27:33: Lp2: Nothing to do yet. Incrementing timer. 77 < 360
2021-01-05 11:27:43: Lp2: Nothing to do yet. Incrementing timer. 78 < 360
2021-01-05 11:27:53: Lp2: Nothing to do yet. Incrementing timer. 79 < 360
2021-01-05 11:28:03: Lp2: Nothing to do yet. Incrementing timer. 80 < 360
2021-01-05 11:28:13: Lp2: Nothing to do yet. Incrementing timer. 81 < 360
2021-01-05 11:28:23: Lp2: Nothing to do yet. Incrementing timer. 82 < 360
2021-01-05 11:28:33: Lp2: Nothing to do yet. Incrementing timer. 83 < 360
2021-01-05 11:28:43: Lp2: Nothing to do yet. Incrementing timer. 84 < 360
2021-01-05 11:28:53: Lp2: Nothing to do yet. Incrementing timer. 85 < 360
2021-01-05 11:29:03: Lp2: Nothing to do yet. Incrementing timer. 86 < 360
2021-01-05 11:29:13: Lp2: Nothing to do yet. Incrementing timer. 87 < 360
2021-01-05 11:29:23: Lp2: Nothing to do yet. Incrementing timer. 88 < 360
2021-01-05 11:29:33: Lp2: Nothing to do yet. Incrementing timer. 89 < 360
2021-01-05 11:29:43: Lp2: Nothing to do yet. Incrementing timer. 90 < 360
2021-01-05 11:29:53: Lp2: Nothing to do yet. Incrementing timer. 91 < 360
2021-01-05 11:30:03: Lp2: Nothing to do yet. Incrementing timer. 92 < 360
2021-01-05 11:30:13: Lp2: Nothing to do yet. Incrementing timer. 93 < 360
2021-01-05 11:30:23: Lp2: Nothing to do yet. Incrementing timer. 94 < 360
2021-01-05 11:30:33: Lp2: Nothing to do yet. Incrementing timer. 95 < 360
2021-01-05 11:30:43: Lp2: Nothing to do yet. Incrementing timer. 96 < 360
2021-01-05 11:30:53: Lp2: Nothing to do yet. Incrementing timer. 97 < 360
2021-01-05 11:31:02: Lp2: Nothing to do yet. Incrementing timer. 98 < 360
2021-01-05 11:31:12: Lp2: Nothing to do yet. Incrementing timer. 99 < 360
2021-01-05 11:31:22: Lp2: Nothing to do yet. Incrementing timer. 100 < 360
2021-01-05 11:31:32: Lp2: Nothing to do yet. Incrementing timer. 101 < 360
2021-01-05 11:31:42: Lp2: Nothing to do yet. Incrementing timer. 102 < 360
2021-01-05 11:31:52: Lp2: Nothing to do yet. Incrementing timer. 103 < 360
2021-01-05 11:32:03: Lp2: Nothing to do yet. Incrementing timer. 104 < 360
2021-01-05 11:32:13: Lp2: Nothing to do yet. Incrementing timer. 105 < 360
2021-01-05 11:32:23: Lp2: Nothing to do yet. Incrementing timer. 106 < 360
2021-01-05 11:32:33: Lp2: Nothing to do yet. Incrementing timer. 107 < 360
2021-01-05 11:32:43: Lp2: Nothing to do yet. Incrementing timer. 108 < 360
2021-01-05 11:32:53: Lp2: Nothing to do yet. Incrementing timer. 109 < 360
2021-01-05 11:33:03: Lp2: Nothing to do yet. Incrementing timer. 110 < 360
2021-01-05 11:33:13: Lp2: Nothing to do yet. Incrementing timer. 111 < 360
2021-01-05 11:33:23: Lp2: Nothing to do yet. Incrementing timer. 112 < 360
2021-01-05 11:33:33: Lp2: Nothing to do yet. Incrementing timer. 113 < 360
2021-01-05 11:33:43: Lp2: Nothing to do yet. Incrementing timer. 114 < 360
2021-01-05 11:33:53: Lp2: Nothing to do yet. Incrementing timer. 115 < 360
2021-01-05 11:34:02: Lp2: Nothing to do yet. Incrementing timer. 116 < 360
2021-01-05 11:34:12: Lp2: Nothing to do yet. Incrementing timer. 117 < 360
2021-01-05 11:34:22: Lp2: Nothing to do yet. Incrementing timer. 118 < 360
2021-01-05 11:34:32: Lp2: Nothing to do yet. Incrementing timer. 119 < 360
2021-01-05 11:34:42: Lp2: Nothing to do yet. Incrementing timer. 120 < 360
2021-01-05 11:34:52: Lp2: Nothing to do yet. Incrementing timer. 121 < 360
2021-01-05 11:35:03: Lp2: Nothing to do yet. Incrementing timer. 122 < 360
2021-01-05 11:35:12: Lp2: Nothing to do yet. Incrementing timer. 123 < 360
2021-01-05 11:35:22: Lp2: Nothing to do yet. Incrementing timer. 124 < 360
2021-01-05 11:35:32: Lp2: Nothing to do yet. Incrementing timer. 125 < 360
2021-01-05 11:35:42: Lp2: Nothing to do yet. Incrementing timer. 126 < 360
2021-01-05 11:35:52: Lp2: Nothing to do yet. Incrementing timer. 127 < 360
2021-01-05 11:36:02: Lp2: Nothing to do yet. Incrementing timer. 128 < 360
2021-01-05 11:36:13: Lp2: Nothing to do yet. Incrementing timer. 129 < 360
2021-01-05 11:36:23: Lp2: Nothing to do yet. Incrementing timer. 130 < 360
2021-01-05 11:36:33: Lp2: Nothing to do yet. Incrementing timer. 131 < 360
2021-01-05 11:36:43: Lp2: Nothing to do yet. Incrementing timer. 132 < 360
[... 281 lines cut ...]
2021-01-05 12:23:33: Lp2: Nothing to do yet. Incrementing timer. 41 < 360
2021-01-05 12:23:43: Lp2: Nothing to do yet. Incrementing timer. 42 < 360
2021-01-05 12:23:53: Lp2: Nothing to do yet. Incrementing timer. 43 < 360
2021-01-05 12:24:02: Lp2: Nothing to do yet. Incrementing timer. 44 < 360
2021-01-05 12:24:12: Lp2: Nothing to do yet. Incrementing timer. 45 < 360
[... cut ...]
2021-01-05 12:26:32: Lp2: Nothing to do yet. Incrementing timer. 59 < 360
2021-01-05 12:26:42: Lp2: Nothing to do yet. Incrementing timer. 60 < 360
2021-01-05 12:26:52: Lp2: Nothing to do yet. Incrementing timer. 61 < 360
2021-01-05 12:27:02: Lp2: Nothing to do yet. Incrementing timer. 62 < 360
2021-01-05 12:27:12: Lp2: Nothing to do yet. Incrementing timer. 63 < 360
2021-01-05 12:27:22: Lp2: Nothing to do yet. Incrementing timer. 64 < 360
2021-01-05 12:27:32: Lp2: Nothing to do yet. Incrementing timer. 65 < 360

DetMoerk, 2021-01-05:
Wenn ich es richtig sehe, dann hast du um 12:16 Uhr eine manuelle Abfrage gestartet, die keine Fehlermeldung zurück gebracht hat. Hätte erwartet, dass du wenn du die Ansicht im Browser aktualisiert, dann auch einen SoC Wert angezeigt bekommst.

StefanHesse, 2021-01-05:
Das war vielleicht das Drücken des runden Pfeils im Webinterface. Es kam aber kein Wert.
Was sagst Du zu diesem Eintrag?
2021-01-05 12:14:44: Lp2: Acc Token Expired

DetMoerk, 2021-01-05:
Das ist OK. Der Access Token hat eine Lebenszeit von 2h. Dann muss ein neuer abgeholt werden. Das wurde erkannt und der Refresh hat wohl funktioniert, sonst müsste eine Fehlermeldung im Log sein. 

VG
Det

StefanHesse, 2021-01-05:
@DetMoerk 
OK, macht Sinn. Leider wird mir immer noch 0% als SoC-Wert angezeigt.
Kann ich noch was tun, um Dir zu helfen das Problem zu finden? Kann ich irgendwo was kontrollieren?

DetMoerk, 2021-01-05:
@StefanHesse : ich bin gerade etwas am Grübeln, wie ich dahinter komme. 
Es fehlt auf jeden Fall eine Meldung wie diese im 
Log 2021-01-05 12:33:42: Lp1: SOC: 100 RANGE: 49
Ich habe einen Debug Mode, daber den habe ich erst mal fix abgeschaltet, damit das soc.log nicht geflutet wird. 

@benderl: Gibt es nur Debug =1 und debug=0 oder habt ihr mehrere Log Level implementiert, so dass ich unterscheiden könnte?

VG
Det

DetMoerk, 2021-01-05:
@benderl: habe es gerade gefunden. 0, 1, 2 sind möglich. habe das jetzt angepasst, so dass soc.log bei Level 1 und 2 detailierter gefüllt werden.

DetMoerk, 2021-01-05:
@StefanHesse : Ichhabe gerade einen Pull Request eingestellt, damit dr Debug Level dynamisch in der openWB geändert werden kann. Damit kommen wir dann hoffentlich weiter.

Wenn gemerged, dann bitte unter Einstellungen/System/Debugging den Level auf Mode 2 stellen, Testen und das resultierende Log direkt an mich schicken. 

VG
Det

StefanHesse, 2021-01-05:
@DetMoerk 
Der PullRequest ist drin und ich habe gleich ein Update gemacht und den Debug Level auf 2 gestellt. Aber wirklich viel mehr ist da nicht zu sehen im Log:

2021-01-05 18:12:13: Lp2: Nothing to do yet. Incrementing timer. 64 < 360
2021-01-05 18:12:23: Lp2: Nothing to do yet. Incrementing timer. 65 < 360
2021-01-05 18:12:33: Lp2: Nothing to do yet. Incrementing timer. 66 < 360
2021-01-05 18:12:43: Lp2: Nothing to do yet. Incrementing timer. 67 < 360
2021-01-05 18:12:53: Lp2: Nothing to do yet. Incrementing timer. 68 < 360
2021-01-05 18:13:03: Lp2: Nothing to do yet. Incrementing timer. 69 < 360
2021-01-05 18:13:13: Lp2: Nothing to do yet. Incrementing timer. 70 < 360
2021-01-05 18:13:23: Lp2: Nothing to do yet. Incrementing timer. 71 < 360
2021-01-05 18:13:33: Lp2: Requesting SoC
2021-01-05 18:13:35: Lp2: client: 1d794f97-xxxx-xxxx-xxxx-a73053c36217
2021-01-05 18:13:35: Lp2: SOC URL: https://api.mercedes-benz.com/vehicledata/v2/vehicles/FIN/containers/electricvehicle
2021-01-05 18:13:43: Lp2: Nothing to do yet. Incrementing timer. 0 < 360
2021-01-05 18:13:53: Lp2: Nothing to do yet. Incrementing timer. 1 < 360
2021-01-05 18:14:03: Lp2: Nothing to do yet. Incrementing timer. 2 < 360
2021-01-05 18:14:13: Lp2: Nothing to do yet. Incrementing timer. 3 < 360
2021-01-05 18:14:23: Lp2: Nothing to do yet. Incrementing timer. 4 < 360
2021-01-05 18:14:33: Lp2: Nothing to do yet. Incrementing timer. 5 < 360
2021-01-05 18:14:43: Lp2: Nothing to do yet. Incrementing timer. 6 < 360
2021-01-05 18:14:53: Lp2: Nothing to do yet. Incrementing timer. 7 < 360

Gruß,
Stefan

DetMoerk, 2021-01-05:
@StefanHesse : Du warst leider zu schnell. Ich vermute ich hab was verbasselt. Die Version der SoC Ermittlung ist noch nicht die die ich auf meinem Raspi habe, Nächster Request ist schon nachgeschoben.

VG
Det

DetMoerk, 2021-01-05:
@StefanHesse : Aber  wenn ich einen anderen PR richtig gelesen habe kannst du jetzt den Hostname einer Box anpassen ;-).

VG
Det

StefanHesse, 2021-01-05:
@DetMoerk 
Schon ausprobiert und klappt :-)
Endlich liefert ein nslookup nur noch eine IPv6 und eine IPv4-Adresse zurück und nicht mehr beide. Wenn es dann mit dem SoC erst mal klappt, wäre mein nächster Test das dann auf DNS-Namen umzustellen. Aber erst mal teste ich mit der IP-Adresse bis es zuverlässig klappt. Nicht alles auf einmal...

DetMoerk, 2021-01-05:
@StefanHesse :Bitte mal warten mit weiterem Update, momentan ist eine SocVersion drin die sicher nicht funktioniert.

StefanHesse, 2021-01-05:
@DetMoerk 
Die habe ich dann jetzt ja bestimmt auch drauf. Soll ich das SoC Modul erst mal wieder deaktivieren oder kann ich es einfach so lassen?
Nicht dass ich Mercedes mit ungültigen Abfragen flute und die mich sperren oder so :-)

DetMoerk, 2021-01-05:
Deaktiviere mal. Sollte nichts passieren, da die SoC Abfrage direkt auf die Nase fällt und nicht zur Mercedes API geht, aber sicher ist sicher. 

VG
Det

DetMoerk, 2021-01-05:
@StefanHesse : Jetzt ist die korrekte Version drin. 
Du musst auf jeden Fall nach dem Installieren noch mal den Link anklicken.
[image]

Als Response musst du auf diese Seite weiter geleitet werden.
[image]
Wenn das nicht kommt, dann muss ich mich noch mal um die Callback Funktion kümmern. Das hat bei mir bisher immer funktioniert. 

Bitte mal testen und Debug Level auf 2 und dann das soc.log runter laden. Es reicht, nur einen Teil auszuschneiden. Die ganzen Nothing to do yet Zeilen können raus. 

VG
Det

StefanHesse, 2021-01-05:
@DetMoerk 
Kann ich das auch jetzt schon testen, auch wenn @benderl noch nicht commitet hat?

DetMoerk, 2021-01-05:
mein letzter PR ist schon drin. #859

StefanHesse, 2021-01-05:
Leider kein Erfolg. Es kommt jetzt wieder nur ein leeres Fenster und nicht mal mehr die Mercedes-Seite.
Hier das Log:
2021-01-05 18:51:33: Lp2: Nothing to do yet. Incrementing timer. 227 < 360
2021-01-05 19:59:32: Lp2: Requesting SoC
2021-01-05 19:59:34: Lp2: Debug Level: 2
2021-01-05 19:59:34: Lp2: client: 1d794f97-xxxx-xxxx-xxxx-a73053c36217
2021-01-05 19:59:34: Lp2: SOC URL: https://api.mercedes-benz.com/vehicledata/v2/vehicles/FIN/containers/electricvehicle
2021-01-05 19:59:42: Lp2: Nothing to do yet. Incrementing timer. 0 < 360

DetMoerk, 2021-01-05:
Dann muss ich mich mal um den Link kümmern. Das liegt nicht am soc Modul sondern an den callback Handlern. 
Mal schauen, ob ich morgen dazu komme. 
Du kannst das SoC Modul nicht zufällig mal auf LP1 aktivieren und schauen, ob es da geht?

[27 more replies]

closed 2021-01-07 by benderl