snaptec/openWB #877: Initialisierungsfehler bei Phasenumschaltung

mode2k, 2021-01-09:
Software openWB Version 1.9.003 Stable

Konfiguration: 
Sofortladen 3P
PV Laden 1P

Das Problem tritt nur unmittelbar nach einem Start der Box auf und ist reproduzierbar bis Mode PV aktiviert wurde. 

Im Detail:
Nach dem Einschalten sind beim Startvorgang des Raspys beide Pulsausgänge aktiv.
Nicht gut für das Relais, aber nicht zu ändern, da noch keine Kontrolle über GPIOs möglich.
Nach Starten der OpenWB Software bleibt Impulsausgang 1P jedoch dauerhaft aktiv.
Die Box ist nach dem Booten im Mode Stop, während 1P aktiv bleibt.
Wird auf Mode Sofortladen geschaltet, so werden beide Impulsausgänge 1P und 3P dauerhaft aktiv geschaltet. 
Das hat zur Folge dass das Relais nicht umschaltet und weiterhin mit 1P geladen wird. 
Wird wieder auf Mode Stop gewechselt wird Impulsausgang 3P deaktiviert und 1P bleibt weiterhin aktiv.
Dieses Verhalten kann beliebig oft durch Wechseln zwischen Stop und Sofortladen reproduziert werden. 
Es hört erst dann auf, Wenn nach dem Reboot einmal der Mode PV (oder PV min oder ein anderer 1P Mode) aktiviert wurde. 
Ab dann ist 1P nicht mehr dauerhaft aktiv und es funktionieren beide Impulsausgänge wie gewünscht und toggeln das Relais für 1s während die Ladung pausiert ist.
 
Ich sehe dabei konkret folgende Probleme:
- Sofortladen (Konfiguriert als 3P) lädt nach Reboot mit 1P, bis einmal ein 1P Mode aktiviert wurde.
- Die Relaisspulen werden teileweise beide dauerhaft bestromt. Das überlebt das Relais zwar einige Zeit, 
  es ist jedoch nicht dafür konstruiert und wird in einem nicht erlaubten Zustand betrieben.
- Wenn beide Spulen bestromt werden besteht die Gefahr dass das Relais spontan schalten könnte.

All dies habe ich nur mit LP1 getestet, gehe aber davon aus dass es ebenso LP2 betrifft.

snaptec, 2021-01-10:
Sollte nach Neustart gefixt sein.
https://github.com/snaptec/openWB/commit/ebafb2832f734f307a840513a8c20226e58115a6

mode2k, 2021-01-10:
Danke für die schnelle Reaktion.
Leider bleibt Pin 37 weiterhin dauerhaft aktiv bis einmal der PV Mode aktivert wurde.
Fehlerbild hat sich nicht verändert.

Warum hast du zum Start ein kurzes Toggeln vom CP Relais 1 eingefügt?

snaptec, 2021-01-11:
Noch ein Typo, nun gefixt und getestet.
Umschaltung erfolgt immer nur bei getrenntem CP Signal.

closed 2021-01-11 by snaptec

mode2k, 2021-01-11:
Funktioniert nun. Danke!