snaptec/openWB #884: WebHooks beim Beginn und Beenden der Ladung

tomac01, 2021-01-12:
Sachverhalt:
Aktuell kann OpenWB so konfiguriert werden, dass einmalig ein WebHook beim Anschließen des EV an den LP abgesetzt wird. In vielen Fällen (z.B. im Winter) wird bei mir das EV über "Sofort laden" nach dem Einstecken dann auch direkt geladen.
Es bleibt dann aber über längere Zeit angeschlossen. Beginnt zwischendurch dann eine erneute Ladung (z.B. bei Abfall des SoC vom EV), dann wäre es wünschenswert, wenn ein erneuter Webhook abgesetzt werden könnte.

Hintergrund: Ich tracke meine Ladehistorie mit dem Teslalogger. Dieser legt sich nach ca. 60 Min. schlafen, damit mein EV in den Energiesparmodus gehen kann. Wird nun die Ladung wieder wegen unterschreiten des SoC aufgenommen, aktiviert sich das EV aber nicht unbedingt der Teslalogger. Durch das Absetzen des Webhook kann ich diesen ebenfalls aus seinem Schlafmodus holen.

Anforderung:
Die openWB soll die Möglichkeit bereitstellen, zusätzlich bei Beginn und beim Beenden einer EV-Ladung einen Webhook optional absetzen zu können.

Vielen Dank!

seb585, 2021-02-20:
Macht zusätzlich ein konfigurierbarer WebHook beim Abstecken Sinn, oder gibt es dafür keinen Anwendungsfall?

tomac01, 2021-02-20:
Grundsätzlich ja, damit hätte man dann alle Situationen abgedeckt.

tomac01, 2021-03-24:
Funktioniert! Vielen Dank!

closed 2021-03-24 by tomac01