snaptec/openWB #886: VZLogger ermittlung über uuid Werte

DetMoerk, 2021-01-12:
Hallo Team, 
aktuell werden die Watt/Bezug/Einspeisung und PVWatt aus vzlogger über die Zeilenzahl der JSON Antwort ermittelt. Funktioniert soweit problemlos und habe ich aktuell auch im Einsatz. 

[image]

Werden im VZlogger allerdings Kanäle angelegt, werden diese nicht unbedingt am Ende des JSON Reponse angehängt. Daraus ergibt sich die Notwendigkeit die openWB Konfiguration für EVU und PV zu überprüfen bzw. auch anzupassen. 

Ich habe mir das angeschaut und in Python3 ein Testscript erstellt, um die Werte an Hand der UUID der Kanäle zu ermitteln. 
Damit wird das Modul unabhängig von sich ändernden Zeilenzahlen. 

Wenn die Änderung auch für Andere von Interesse ist, würde ich ein EVU und ein PV Modul (z.B. vzloggeruuid, vzloggeruuidpv) schreiben (python3). Ich würde dazu neue Konfig Variablen definieren und das Modul als Alternative zum bisherigen vzlogger/vzloggerpv anbieten, sodass das Modul keine bestehenden und funktionierenden Konfigurationen zerstört.

VG
Det

benderl, 2021-01-12:
Das klingt gut. Kannst Du Dir eine automatische Migration von Zeilen zu UUID vorstellen? Zwei Module erzeugen auch immer den doppelten Pflege- und Support-Aufwand.
Beim Aufruf der main.sh prüfen, ob in der Variablen eine Zeilennummer oder UUID vorhanden ist. Bei Zeilennummern die alte Logik anwenden und die UUID der Zeile parsen und in der Konfiguration ersetzen.
Das sollte doch funktionieren.

DetMoerk, 2021-01-12:
Ja kann man theoretisch auch machen, Risiko wäre, dass jeder Eintrag einzeln geprüft werden muss. 
Auf jeden Fall muss dann bei Zeilen immer der curl |jq . Aus der bash verwendet werden, damit die Formatierung auch zwingend identisch mit dem Test Aufruf zur Ermittlung der Zeilen ist. Nur bei uuid geht es dann in das Python Script. 

Auf jeden Fall ein wesentlich höheres Fehlerrisiko. Hoffe es gibt genügend Tester mit openWB.

VG
Det

DetMoerk, 2021-01-17:
@benderl hab das gerade noch mal gelesen was du geschrieben hast. 
Habs wohl beim ersten Mal nicht richtig kapiert. Du willst, dass ich automatisch die uuid zur eingegebene Zeile ermittle, und diese dann in der Konfiguration hinterlege. 
Solche "background magic" finde ich immer riskant. 
Das müsste wirklich gut und ausgiebig getestet werden, damit das wirklich wasserdicht ist. Ich hab keinen Plan wie verbreitet das Modul ist und wie auch die QS bei euch läuft. 
Aber mit sowas kann man sich auch ganz schön in die Nesseln setzen und den guten Ruf aufs Spiel setzen. Da fände ich eine Roadmap un deprecated Module besser. Also neues Modul mit einbauen und die Leute auffordern ihre Konfiguration entsprechend zu überführen und zu testen. Nach einer gewissen Übergangsphase wird das alte Modul raus genommen.
Zuerst hatte ich verstanden, dass ich schaue ob in der Konfiguration eine Zeile (positiv, numerisch, <=1000) ist. Wenn ja, dann mit dem Bash Programm weiter, wenn Nein dann wird angenommen, dass es eine uuid ist und es wird ein python Script aufgerufen, dass den entsprechenden Wert extrahiert.

VG
Det

DetMoerk, 2021-01-18:
Wenn ich die Zeilen im Hintergrund durch UUIDs ersetzen würde, wären auch Probleme vorprogrammiert, wenn ein VZ User von der Nightly wieder auf eine Stable zurück wechselt.

VG
Det

DetMoerk, 2021-02-02:
Komme gerade nicht zum Umbauen. Hab es aber noch um Hinterkopf.

DetMoerk, 2021-02-06:
@benderl : Es gibt im Forum ja eine Anfrage, ob der Faktor von Meeswandlerzählern in openWB eingestellt werden kann. Soll ich da wenn ich die Änderung in Richtung UUID baue, auch einen Faktor vorsehen?

benderl, 2021-02-06:
Wenn Du willst, mach das ruhig. Den Fehler sehe ich aber eindeutig im VZLogger.

DetMoerk, 2021-02-06:
Ich versuche gerade mit dem Anforderer zu klären, wo überall der Faktor einzuführen ist. 
Wenn ich das richtig verstanden habe, hängt das bei ihm mit der Erfassung durch einen Messwandlerzähler zusammen. Er hat einen Anschluß >150kW und auch die Anzeige auf seinem Zähler / Smartmeter ist ebenfalls Faktor 50 zu klein und sein EVU rechnet das einfach hoch um die Rechnung zu schreiben. 
Frage ist halt, wie oft kommt die Konstellation vor, dass jemand der einen Firmenanschluss hat, mit VZ den EVU speist. 

> Wobei gerade im gewerblichen Bereich das sehr interessant ist. Vor allem mit vielen Panels auf dem Dach und einer EV Flotte.
[https://www.photovoltaikforum.com/thread/151583-emh-ebz-2-wege-messwandlerz%C3%A4hler-in-vzlogger/?pageNo=3](PV Forum)

DetMoerk, 2021-02-06:
@benderl : Der VZlogger PV hat nur eine Ermittlung der aktuellen Leistung. Wenn ich mir das PV JSON Modul anschaue kann ich da auch den Zählerstand ermitteln lassen. Ich würde das auch in das VZloggerPV Modul einbauen. Dann kann da auch der Zählerstand des Einspeisezählers in openWB verwendet werden.

DetMoerk, 2021-02-07:
@benderl: Habe mir heute noch mal die JSON Module angeschaut und auch eine Möglichkeit gefunden, die etwas komplexere Ausgabe des VZLoggers zu parsen. Mit einer kleinen Änderung könnte man das VZLogger Modul abdecken, sogar die Multiplikation für einen Messwandlerzähler kann direkt im passenden jq Filter Statement eingegeben werden. 
Hat etwas gedauert, bis ich dahinter gekommen bin, warum es nicht funktioniert hat. Daher habe ich auch zusätzliches Logging eingebaut. Ich würde das nacher mal als Pull Request noch einstellen. Da müsste man aber evtl. noch mal drüber schauen. 

VG
Det

DetMoerk, 2021-02-07:
#1012 kann die Anforderung abdecken, VZLogger zu ersetzen. Entsprechendes Filter Statement: 
.data[] | select(.uuid=="<UUID des entsprechenden Wertes>")| .tuples[0][1]

benderl, 2021-02-08:
@DetMoerk Danke für Deinen PR!
Das klingt vielversprechend. Siehst Du potentiell Probleme durch das Update des JSON Moduls? Die ergänzten Doublequotes um den JQ String sollten meiner Einschätzung nach keine Auswirkungen auf bestehende, einfache Selektoren haben.

DetMoerk, 2021-02-08:
Ich kann es heute Abend noch etwas ausführlicher testen, aber ich denke dass es keine Probleme mit einfachen Filtern gibt. Hätte eher erwartet, dass es davor schon mal Dinge gab, für die die Filter nicht funktioniert haben.

Ich hab leider keine Ahnung welche Module wie oft eingesetzt werden. Ich vermute, dass ihr da auch keine Details kennt. So lange es keine Issues damit gibt, kommt wohl auch keine Rückmeldung.

Wie ist denn deine Erfahrung mit Nightly und Erwartungen an die Stabilität

VG
Det

benderl, 2021-02-08:
Naja, auf der Update-Seite steht folgendes klar und deutlich:
"Die Nightly-Version beinhaltet Neuentwicklungen, die teils nur eingeschränkt getestet sind. Fehlverhalten ist wahrscheinlich."
Demnach kann ich das problemlos in die Nightly aufnehmen. ;-)
Selbst für die Beta steht dort "Fehlverhalten ist nicht ausgeschlossen."
Wenn Du das heute nochmal testen kannst und mir grünes Licht gibst, kommt es rein. Wir wollen unbedingt zügig in Richtung Beta und dann auch Stable, da inzwischen doch einige Fehler in der Stable behoben wurden, die immer wieder für Probleme sorgen. viele Endanwender (nicht die "Nerds") bleiben lieber auf Stable.

DetMoerk, 2021-02-08:
Alles klar. Ich hab heute noch Termine bis 17:00 Uhr, Sa wo ich mein Geld verdiene😉. Ich teste noch mal ein paar Dinge. 
Aber ich denke das JSON Modul ist eher bei den Nerds eingesetzt. 

VG
Det

DetMoerk, 2021-02-08:
@benderl : Das mit dem zusätzlichen Logging passt oder soll ich das wieder raus nehmen? Wie ist da eure Strategie?

Wenn alles klappt, nehme ich natürlich die fixe Debug Level Zuordnung noch weg. 

VG
Det

benderl, 2021-02-08:
Vom Prinzip her bin ich bei Dir, analog zu den SoC-Modulen auch hier ein separates Logfile zu nutzen. Aktuell würde ich das aber noch weglassen und in das normale Logfile schreiben, damit die Ausgaben auch in den Tickets für den Support enthalten ist.

DetMoerk, 2021-02-08:
@benderl : bedeutet einfach ins openwb.log in der Ramdisk appenden?

VG
Det

benderl, 2021-02-08:
Ja, genau so.

DetMoerk, 2021-02-08:
@benderl: Ichhab noch mal getestet, das sieht auch z.B. mit dem dargestellten Beispiel in derKofig des EVU Moduls gut aus. Von daher sollten keine Seiteneffekte auftreten. Logging habe ich auf openWB.log angepasst und einen Modul Kenner EVU/PV zusätzlich eingefügt. Ausgabe in openWB.log nur bei Debug >0

benderl, 2021-02-08:
Danke für Deinen Beitrag und die Geduld mit uns.

closed 2021-02-08 by benderl