snaptec/openWB #887: Graph in der Übersicht nicht gleichmäßig

Canonip, 2021-01-13:
In der Übersicht sind die Bezeichnungen der zeitlichen Abstände auf der X-Achse nicht gleichmäßig.
z.B. 11:44 - 11:53 (9 Minuten) und danach 11:54 (1 Minute)

[image]

Einstellungen:

- 30 Minuten Zeitintervall im Graph
- Standard Theme
- PV+EVU+Speicher eingebunden (Kostal Plenticore)
- OpenWB Version 1.9.009 (2021-01-08 12:33:35 +0100 [0bf3001c] ) auf OpenWB series2

snaptec, 2021-01-13:
War der Browser Tab stellenweise im Hintergrund?
Passt es nach einem Reload der Seite?

Canonip, 2021-01-13:
Ausprobiert mit Chrome 87, Edge 87, Firefox 84 auf Windows 10 1909
Neustart der Browser hat nicht geholfen
Reboot (und 30 min warten) der openWB hat nicht geholfen.

Ich wechsle gleich von Nightly 1.9.009 zu Stable 1.9.003 und schaue ob es auch hier der Fall ist

Canonip, 2021-01-13:
Bei 1.9.003 das gleiche:
[image]

cshagen, 2021-01-29:
Es handelt sich um einen Fehler in graphing.sh.
Die Zeilen 50-99 in all-live.graph auf der Ramdisk werden nicht per MQTT publiziert. 
Ich habe das bei mir lokal korrigiert und es funktioniert. Kann einen Pull-Request einstellen, falls gewünscht.

benderl, 2021-01-29:
@cshagen ja, bitte mach das.

benderl, 2021-01-30:
Sollte mit #983 behoben sein.

closed 2021-01-30 by benderl