snaptec/openWB #889: Keine Code Eingabe auf Display in Nightly

DetMoerk, 2021-01-13:
Hallo zusammen,

in der Nightly ist auf dem Display der Standard+ keine Eingabe für den Code mehr vorhanden . Es gibt jetzt einen Punkt Status, aber die Code Eingabe finde ich nicht mehr.

VG
Det

DetMoerk, 2021-01-13:
[image]

[image]

benderl, 2021-01-13:
Da das quasi ein manueller Rfid ist, wird der Punkt nur noch angezeigt, wenn auch Rfid im Modus 1 oder 2 aktiviert ist.

DetMoerk, 2021-01-13:
Jetzt muss ich dann doch mal fragen, ob sowas irgendwo dokumentiert ist.

VG
Det

benderl, 2021-01-14:
https://github.com/snaptec/openWB/commit/4efac89f924ca8c2a98f254a88ec0f9ef20cda15

closed 2021-01-14 by benderl