snaptec/openWB #974: Nissan Leaf SOC funktioniert nicht

bubbagump24, 2021-01-27:
Die Abfrage des SOC für Nissan Leaf Modell ZE1 (ab 2019) funktioniert derzeit nicht. Basierend auf der Konversation unter https://openwb.de/forum/viewtopic.php?f=5&t=432&start=50 wäre die Lösung in evcc wohl schon vorhanden.

Das SOC Log zeigt folgendes:

`2021-01-27 23:16:41,029 - pycarwings2 - ERROR - carwings error INVALID PARAMS: -2010
2021-01-27 23:16:41,028 - pycarwings2 - DEBUG - Response HTTP Response Body: {"status":"-2010","message":"INVALID PARAMS","resultKey":""}
2021-01-27 23:16:41,027 - pycarwings2 - DEBUG - Response HTTP Status Code: 200
2021-01-27 23:16:41,021 - requests.packages.urllib3.connectionpool - DEBUG - https://gdcportalgw.its-mo.com:443 "POST /api_v190426_NE/gdc/UserLoginRequest.php HTTP/1.1" 200 None
2021-01-27 23:16:37,739 - requests.packages.urllib3.connectionpool - DEBUG - Starting new HTTPS connection (1): gdcportalgw.its-mo.com
}
   "initial_app_str": "9s5rfKVuMrT03RtzajWNcA"
   "custom_sessionid": "",
   "UserId": "xxxxxxxx@gmail.com",
   "RegionCode": "NE",
   "Password": "xxxxxxxxxxx",
2021-01-27 23:16:37,737 - pycarwings2 - DEBUG - params: {
2021-01-27 23:16:37,737 - pycarwings2 - DEBUG - invoking carwings API: https://gdcportalgw.its-mo.com/api_v190426_NE/gdc/UserLoginRequest.php
2021-01-27 23:16:37,729 - pycarwings2 - DEBUG - Response HTTP Response Body: {"status":200,"message":"success","baseprm":"88dSp7wWnV3bvv9Z88zEwg"}
2021-01-27 23:16:37,729 - pycarwings2 - DEBUG - Response HTTP Status Code: 200
2021-01-27 23:16:37,724 - requests.packages.urllib3.connectionpool - DEBUG - https://gdcportalgw.its-mo.com:443 "POST /api_v190426_NE/gdc/InitialApp_v2.php HTTP/1.1" 200 None
2021-01-27 23:16:36,606 - requests.packages.urllib3.connectionpool - DEBUG - Starting new HTTPS connection (1): gdcportalgw.its-mo.com
}
   "lg": "en-US"
   "initial_app_str": "9s5rfKVuMrT03RtzajWNcA",
   "custom_sessionid": "",
   "RegionCode": "NE",
2021-01-27 23:16:36,603 - pycarwings2 - DEBUG - params: {
2021-01-27 23:16:36,602 - pycarwings2 - DEBUG - invoking carwings API: https://gdcportalgw.its-mo.com/api_v190426_NE/gdc/InitialApp_v2.php
2021-01-27 23:16:36,586 - root - DEBUG - LP1: login = xxxxxxxxx@gmail.com, region = NE
2021-01-27 23:16:36,585 - root - DEBUG - LP1: Update soctimer to 10`

andig, 2021-01-31:
> Die Abfrage des SOC für Nissan Leaf Modell ZE1 (ab 2019) funktioniert derzeit nicht. Basierend auf der Konversation unter https://openwb.de/forum/viewtopic.php?f=5&t=432&start=50 wäre die Lösung in evcc wohl schon vorhanden.

@bubbagump24 Du kannst das mittels evcc einfach ausprobieren. Auto in der Konfigurationsdatei anlegen und dann einfach

    evcc vehicle -c <config>

zum testen.

bubbagump24, 2021-01-31:
Hi andig, danke für die Antwort. Ich habe eine Openwb Series2 Standard+, d.
h. keine Möglichkeit auf den Raspi dahinter per SSH zu gelangen. Wenn ich
das richtig verstehe, wäre evcc eine separat zu installieren Applikation,
oder?
Klar könnte ich einen separaten Raspi aufsetzen, aber eigentlich wäre das
Ziel dieses issues das der OpenWB beizubringen.

Alex-ander-s, 2021-02-07:
Hi,
also ich hatte immer wieder Probleme, wenn kein User-Agent im Request zu den Nissan Servern übermittelt wird.
Das hat in meinem Fall geholfen den "INVALID PARAMS" loszuwerden:
==========
--- a/pycarwings2.py
+++ b/pycarwings2.py
@@ -118,7 +118,12 @@ class Session(object):
         else:
             params["custom_sessionid"] = ""
 
-        req = Request('POST', url=BASE_URL + endpoint, data=params).prepare()
+        req = Request(
+            'POST',
+            url=BASE_URL + endpoint,
+            data=params,
+            headers={"User-Agent": ""}
+        ).prepare()
 
         log.debug("invoking carwings API: %s" % req.url)
         log.debug("params: %s" % json.dumps(
==========

Alex-ander-s, 2021-02-07:
Hi,
also ich hatte immer wieder Probleme, wenn kein User-Agent im Request zu den Nissan Servern übermittelt wird.
Das hat in meinem Fall geholfen den INVALID PARAMS loszuwerden:
`--- a/pycarwings2.py
+++ b/pycarwings2.py
@@ -118,7 +118,12 @@ class Session(object):
         else:
             params["custom_sessionid"] = ""
 
-        req = Request('POST', url=BASE_URL + endpoint, data=params).prepare()
+        req = Request(
+            'POST',
+            url=BASE_URL + endpoint,
+            data=params,
+            headers={"User-Agent": ""}
+        ).prepare()
 
         log.debug("invoking carwings API: %s" % req.url)
         log.debug("params: %s" % json.dumps(`

Alex-ander-s, 2021-02-07:
Hi,
also in meinem Fall hatte ich Probleme, wenn kein User-Agent zu den Nissan Servern übermittelt wird.
Das hat geholfen den "INVALID PARAMS" loszuwerden:

[pycarwings2.py.patch.txt](https://github.com/snaptec/openWB/files/5938676/pycarwings2.py.patch.txt)

andig, 2021-02-07:
Bei meinen Tests ging es ohne User-Agent. @bubbagump24 kannst Du das Auto denn von einem anderen Rechner aus (muss ja nicht die OpenWB sein) mit 

    evcc vehicle -c <deine config>

abfragen? Details zur Konfiguration gibts bei evcc.

andig, 2021-03-09:
ping @bubbagump24 @Alex-ander-s gibts da noch Probleme mit evcc-soc?

Alex-ander-s, 2021-03-18:
@andig Sorry, hat ein wenig gedauert.
Ich nutze üblicherweise nicht die in evcc integrierte SOC Abfrage, sondern meinen eigenen fork von pycarwings2 mit dem ich den SOC in eine Datenbank schreibe.
Ich habe testweise mal versucht das Nissan Modul in evcc selbst zu konfigurieren, bin aber leider gescheitert (mehr als HTTP 401 vom Nissan Server bekomme ich nicht).
Mit meinem python Script und den selben Zugangsdaten geht es jedenfalls.

bubbagump24, 2021-03-19:
@andig Ich habe es nun endlich mal geschafft evcc auf den Raspi zu packen und siehe da: Funktioniert! 
Zumindest bekomme ich den SOC und die Reichweite ausgelesen. Außerdem bekomme ich Werte von meinem Sunny Home Mangager als EVU Zähler und die openwb wird auch erkannt. 
Mir ist noch nicht klar wie ich evcc beibringe mit der openwb bilateral zu kommunizieren, denn bisher wird als Ladeleistung der Wert des EVU Meters angezeigt und ich kann keinen Befehl an die openwb schicken. Auch das Auslesen des SMA STP10 Wechselrichters via Modbus will noch nicht ("ist kein sunspec device"), aber morgen ist ja auch noch ein Tag. 

Das grundlegende Problem des SOC ist jedenfalls gelöst, hurra! Wie bekommen wir das nun an die OpenWB?

andig, 2021-03-20:
> Das grundlegende Problem des SOC ist jedenfalls gelöst, hurra! Wie bekommen wir das nun an die OpenWB?

wenn Du evcc Nutzung mit openWB meinst würde ich vorschlagen das im evcc Repository zu diskutieren. Ist hier doch arg off-topic ;)

closed 2021-04-07 by benderl

bubbagump24, 2021-04-07:
Hi, ich bin gerade irritiert, da das Thema ohne Lösung und ohne Kommentar geschlossen wurde. Wie geht's jetzt weiter? Gibt es eine Lösung, die hier referenziert werden kann?

benderl, 2021-04-07:
Es gibt mit EVCC (in ein paar Tagen) und TRONITY jetzt zwei Module, die auch den Leaf unterstützen. Leider sind beide mit laufenden Kosten verbunden. Für eine Lösung direkt in openWB haben sich keine Entwickler gemeldet.

andig, 2021-04-10:
@bubbagump24 siehe Diskussion in https://openwb.de/forum/viewtopic.php?f=11&t=2841&p=30912#p30912. Alternativ kannst Du auch mal evcc "standalone" probieren, dann kann ich auch direkt debuggen falls etwas nicht funktioniert.