snaptec/openWB #991: Fehlende Fehlerbehandlung in cron5min

MiOrt, 2021-01-31:
Ich hatte vor Urzeiten im pi das WLAN selbst deaktiviert mittels
`# Disable Wifi`
`dtoverlay=pi3-disable-wifi`
in der `/boot/config.txt`.

Nun habe ich gesehen, dass folgende Fehlermeldung mir das Log vollschreibt:

`SIOCSIFADDR: Kein passendes Gerät gefunden`
`wlan0:0: ERROR while getting interface flags: Kein passendes Gerät gefunden`
`SIOCSIFNETMASK: Kein passendes Gerät gefunden`
`wlan0:0: ERROR while getting interface flags: Kein passendes Gerät gefunden`
`/var/www/html/openWB/runs/cron5min.sh: Zeile 249: /sys/class/net/wlan0/carrier: Datei oder Verzeichnis nicht gefunden`

Ich denke, das liegt an der bereits erfolgten Deaktivierung?!? Hier wäre es schön, wenn eine Fehlerbehandlung diese Log-Einträge verhindert.

snaptec, 2021-02-11:
Liegt daran das openWB das Interface runternimmt sobald LAN vorhanden ist.
Aber als Fallback ohne LAN wird der WLAN Hotspot dann auch wieder gestartet.
Das WLAN komplett zu deaktivieren wie hier passiert ist nicht vorgesehen.

closed 2021-02-11 by snaptec

MiOrt, 2021-02-11:
Es ging auch eher darum, dass ein möglicher Fehler (wie auch immer dieser zustande kommt) dazu führen kann, dass unbemerkt tausende Zeilen ins Logfile geschrieben werden. Dies könnte bestimmt abgefangen werden, wobei mir die Kenntnis dazu im Bereich dieser Skriptsprache fehlt

benderl, 2021-02-11:
Die Meldungen sind bei einer Box ohne Änderungen aber wichtig bei einer Fehlersuche.