snaptec/openWB #1038: aWattar char shows UNIX-time on x-axis

olieske, 2021-02-10:
time is still displayed as HH:MM  on my test installation 1.9.011a (2021-01-31 12:08:39 +0100 [a91651ca]
UNIX-time is  displayed on 1.9.012 (2021-02-10 08:34:53 +0100 [c6ab0b7f]

MiOrt, 2021-02-10:
There is the old js-file in the cache... working on it

MiOrt, 2021-02-10:
It seems that Safari for some reason is ignoring the reload mechanism by using query strings. I could solve the problem by emptying the cache and then completely shut down safari and restart again. What Browser are you on?

olieske, 2021-02-10:
I'm running Firefox on Windows

MiOrt, 2021-02-10:
Strange... I never had this Problem before since we started using query strings. Could you please try to empty the browser Cache and reload the page again, maybe completely restart the browser?

olieske, 2021-02-10:
cleaning the cache fixes the problem.

snaptec, 2021-02-11:
https://github.com/snaptec/openWB/pull/1039

Fixed

closed 2021-02-11 by snaptec