snaptec/openWB #1040: EVU_JSON Default Werte wenn kein Filter eingegeben wird

DetMoerk, 2021-02-10:
Im EVU_JSON Modul (und auch in den Modulen PV_JSON und SPEICHER_JSON 
werden mehrere Werte ermittelt.
z.B. Watt, Bezug kWh, Einspeisung kWh. 
Für die Ermittlung wird eine Abfrage URL und für jeden der Werte ein entsprechendes Filter verwendet.
Wird kein Filter Ausdruck für einen der Werte eingegeben, wird die komplette JSON antwort in die entsprechende Datei auf der ramdisk geschrieben. 
Wären sinnvolle Werte für 
bezugkwh 
einspeisungkwh
pvkwh
speichersoc
jeweils 0 für den Fall, dass keine Werte im JSON vorhanden sind bzw. keine Filter eingegeben werden?

VG
Det

VG
Det

DetMoerk, 2021-02-10:
[image]
@benderl: hast du eine Idee, weshalb der Wert bei JSON Abfrage für kWh abgeschnitten ist? Die Variable wird richtig gespeichert, aber nicht mehr richtig angezeigt.

VG
Det

DetMoerk, 2021-02-10:
Das Problem mit den abgeschnittenen Werten habe ich gefunden. PR wird gestellt.

benderl, 2021-02-10:
In der Ramdisk sollten nur Zahlen ankommen. Bei fehlerhaften Abfragen macht eine "0" Sinn.
Schau Dir auch mal die MQTT Module an. Wir haben jeweils zwei neue Topics eingebaut, um einen Status (0=Ok, 1=Warnung, 2=Fehler) und entsprechenden Text auszugeben. Das kommt noch in die Statusseite.

DetMoerk, 2021-02-10:
Ich habe die drei JSON Module entsprechend angepasst, dass bei den zusätzlichen Werten immer eine 0 eingetragen wird. Die Wattzahlen sind ein Muss, da. PR kommt gleich. Ob ich da mit MQTT heute noch schaffe weiß ich nicht.

VG
Det

benderl, 2021-02-10:
MQTT ist auch kein Muss. Es dauert noch etwas, bis das im Status auftaucht.

closed 2021-02-10 by benderl

DetMoerk, 2021-02-10:
@benderl: was genau meinst du denn mit mqtt module.  Hab gerade ein paar durch geschaut, aber ich habs nicht kapiert.
Oder meinst du den 

openWBmodulepublishState?

VG
Det

benderl, 2021-02-10:
Genau die Funktion meine ich. Da die Module keine Daten abrufen, können die auch keinen Fehler melden. Ist aber wirklich nicht dringend. Plan ist, bei Status 1 oder 2 in der Gui entsprechende Hinweise anzuzeigen. Dann sieht man gleich, ob etwas nicht funktioniert ohne in die Logdateien schauen zu müssen.

DetMoerk, 2021-02-10:
Ich schau mal ob ich da ohne weitere Erklärung weiß was ihr vor habt. Wäre das auch für das SoC Modul, wenn da ein Fehler auftritt? Wäre ja auch im Hintergrund und ansonsten nur über das Logfile zu erkennen.

VG
Det

benderl, 2021-02-11:
Das ist für alle Module geplant. Gerade bei SoC gibt es ja häufig Änderungen und dann funktioniert es nicht mehr.

DetMoerk, 2021-02-11:
Hast du ne Idee, wie ich das in Python Programmen realisiere?
Gibt es da auch nen einfachen Aufruf oder baut ihr da ne Lib?

VG
Det

benderl, 2021-02-11:
Auf die Hilfsfunktion kannst Du so nicht zugreifen. Du könntest den mqtt client nutzen. Bestes Beispiel ist die runs/mqttsub.py

DetMoerk, 2021-02-11:
Ich hab noch mal nachgedacht, bei Fehlern gehe ich mit exit(1) aus dem python Programm raus. Ich muss mal schauen, normalerweise sollte das im entsprechenden main.sh als Return Code ankommen, abhängig davon kann ich dann 

openWBmodulepublishState 

setzen solange es weiter so ist, dass immer ein main.sh als Auslöser aufgerufen wird.
Falls ihr da was in V2 ändert wäre es schön, das davor zu wissen. 

VG
Det

DetMoerk, 2021-02-11:
ich hab es im soc_eq mal eingebaut. Bei den helperfunctions habe ich ein allgemeines openwbDebugLog gesehen. 
Aber wenn ich es richtig gesehen habe, hast du erst in den letzten Tagen auch lokale DebugLog Funktionen in Module eingebaut. 
sind die Helperfunctions neu?
VG
Det

benderl, 2021-02-11:
Ja, die HelperFunctions sind neu, weil ich keine Lust hatte, das in allen anderen Modulen auch einzubauen und zu pflegen. Entwickler sind von Natur aus faule Menschen. ;-) Solange es eine main.sh gibt, stehen die Funktionen dort zur Verfügung, da die aufrufende regel.sh diese vererbt.
Es gibt den MQTT-Explorer, mit dem man sich auf die openWB, genauer den dort laufenden mosquitto broker, verbinden kann. Alles was unter `openWB/set/*` reinkommt, wird jedoch direkt nach der Verarbeitung durch die mqttsub.py wieder gelöscht. Du kannst aber die verarbeiteten Topics unter z. B. `openWB/pv/1/faultState` sehen. Alle entgegengenommenen Nachrichten tauchen auch im mqtt.log auf.
Wenn Du Dich damit etwas beschäftigen möchtest, gerne! Aber bis jetzt ist nur der Grundstein gelegt, indem die Topics definiert wurden. Die Anzeige in der Gui fehlt noch. Vielleicht wartest Du auch noch etwas, bis der Status das anzeigt.

DetMoerk, 2021-02-11:
Dann stelle ich die Module an denen ich gefummelt habe auf die zentrale debuglog um wenn ich ein bisschen Zeit habe.

Meine openWB liegt noch im Arbeitszimmer. Letzte Woche hab ich das Installationsmaterial geholt aber jetzt ist es mir zu kalt um in der Garage zu arbeiten.

Mqtt schaue ich mir an. Hatte ich auch schon mal die ersten Versuche gemacht, aber noch keine richtige Anwendung gehabt. 

VG
Det