snaptec/openWB #1055: ladelog.sh Zeile 202 syntax

Bucky2k, 2021-02-13:
Log entry:

./ladelog.sh: Zeile 202: ((: -: Syntax Fehler: Operator erwartet. (Fehlerverursachendes Zeichen ist \"-\").

snaptec, 2021-02-13:
Was für ein Ladepunkt wird an LP2 genutzt?

Bucky2k, 2021-02-13:
OpenWB Satellit, SDM630 + DINEVSE + Modbus Relaiskarte für 1p3p an einem
Elfin EE11.

Gruß

Sascha

Bucky2k, 2021-02-13:
Aber mein LOG ist eh nicht "clean" sporadische Modbusfehler. Evtl. hängt es
damit zusammen. Dann Issue einfach löschen, ich wollte die Beobachtung nur
mitteilen.

Gruß

Sascha

LOG:

AttributeError: 'ModbusIOException' object has no attribute 'registers'
    lla2 = float(struct.unpack('>f',struct.pack('>HH',*resp.registers))[0])
  File "/var/www/html/openWB/modules/mpm3pmll/readmpm3pm.py", line
105, in <module>
Traceback (most recent call last):
2021-02-13 13:39:53: Modbus EVSE read CP1 issue - using previous state '1' (LV0)
AttributeError: 'ModbusIOException' object has no attribute 'registers'
    print(rq.registers[0])
  File "runs/readmodbus.py", line 21, in <module>
Traceback (most recent call last):
AttributeError: 'ModbusIOException' object has no attribute 'registers'
    llkwh = struct.unpack('>f',struct.pack('>HH',*resp.registers))[0]
  File "/var/www/html/openWB/modules/mpm3pmlllp2/readmpm3pm.py", line
111, in <module>
Traceback (most recent call last):
2021-02-13 11:46:03: Modbus EVSE read CP1 issue - using previous state '1' (LV0)
AttributeError: 'ModbusIOException' object has no attribute 'registers'
    print(rq.registers[0])
  File "runs/readmodbus.py", line 21, in <module>
Traceback (most recent call last):
./ladelog.sh: Zeile 202: ((: -: Syntax Fehler: Operator erwartet.
(Fehlerverursachendes Zeichen ist \"-\").
./ladelog.sh: Zeile 202: ((: -: Syntax Fehler: Operator erwartet.
(Fehlerverursachendes Zeichen ist \"-\").
./ladelog.sh: Zeile 202: ((: -: Syntax Fehler: Operator erwartet.
(Fehlerverursachendes Zeichen ist \"-\").
./ladelog.sh: Zeile 202: ((: -: Syntax Fehler: Operator erwartet.
(Fehlerverursachendes Zeichen ist \"-\").
./ladelog.sh: Zeile 202: ((: -: Syntax Fehler: Operator erwartet.
(Fehlerverursachendes Zeichen ist \"-\").
./ladelog.sh: Zeile 202: ((: -: Syntax Fehler: Operator erwartet.
(Fehlerverursachendes Zeichen ist \"-\").
./ladelog.sh: Zeile 202: ((: -: Syntax Fehler: Operator erwartet.
(Fehlerverursachendes Zeichen ist \"-\").
./ladelog.sh: Zeile 202: ((: -: Syntax Fehler: Operator erwartet.
(Fehlerverursachendes Zeichen ist \"-\").
./ladelog.sh: Zeile 202: ((: -: Syntax Fehler: Operator erwartet.
(Fehlerverursachendes Zeichen ist \"-\").
2021-02-13 10:44:33: Modbus EVSE read CP1 issue - using previous state '1' (LV0)
AttributeError: 'ModbusIOException' object has no attribute 'registers'
    print(rq.registers[0])
  File "runs/readmodbus.py", line 21, in <module>
Traceback (most recent call last):
2021-02-13 09:10:46: IP EVSE read CP2 issue - using previous state '1' (LV0)
AttributeError: 'ModbusIOException' object has no attribute 'registers'
    print(rq.registers[0])
  File "runs/readipmodbus.py", line 23, in <module>
Traceback (most recent call last):
2021-02-13 08:34:56: IP EVSE read CP2 issue - using previous state '1' (LV0)
AttributeError: 'ModbusIOException' object has no attribute 'registers'
    print(rq.registers[0])
  File "runs/readipmodbus.py", line 23, in <module>
Traceback (most recent call last):
2021-02-13 08:19:16: IP EVSE read CP2 issue - using previous state '1' (LV0)
AttributeError: 'ModbusIOException' object has no attribute 'registers'
    print(rq.registers[0])
  File "runs/readipmodbus.py", line 23, in <module>
Traceback (most recent call last):
2021-02-13 07:30:56: IP EVSE read CP2 issue - using previous state '1' (LV0)
AttributeError: 'ModbusIOException' object has no attribute 'registers'
    print(rq.registers[0])
  File "runs/readipmodbus.py", line 23, in <module>
Traceback (most recent call last):
2021-02-13 06:11:33: Modbus EVSE read CP1 issue - using previous state '1' (LV0)
AttributeError: 'ModbusIOException' object has no attribute 'registers'
    print(rq.registers[0])
  File "runs/readmodbus.py", line 21, in <module>

PR #1216 (2021-03-22): fix init some ramdisk files
- should fix #1055
- remove empty backup file
- keep configured current for instant charging
closed 2021-03-22 by benderl