snaptec/openWB #1065: SoC wird nach anstecken gelesen, erscheint aber nicht im Webinterface

martin-zh, 2021-02-16:
Mir fällt auf, dass der SoC nach dem Anstecken zwar korrekt ausgelesen, aber auf der Webseite nicht korrekt angezeigt wird:
Aus dem Log:
2021-02-15 12:01:17: Lp1: SoC: 58
2021-02-15 12:01:17: Lp1: State: "online"
2021-02-15 12:01:11: Lp1: Update SoC
2021-02-15 12:01:11: Lp1: Car state after wakeup: "online"
2021-02-15 12:01:05: Lp1: Waking up car.
2021-02-15 12:01:05: Lp1: Requesting SoC
2021-02-15 05:06:40: Lp1: SoC: 71
2021-02-15 05:06:40: Lp1: State: "online"
2021-02-15 05:06:33: Lp1: Update SoC
2021-02-15 05:06:33: Lp1: Car state after wakeup: "online"
2021-02-15 05:06:26: Lp1: Waking up car.

Anzeige um 13:36 Uhr: Siehe attachment

Ob das nur ein Anzeigefehler ist (dh ob ein Ladevorgang gestartet werden würde ohne manuellen refresh des SoCs) kann ich mangelnds PV Leistung derzeit leider nicht testen.

[image]

benderl, 2021-02-16:
Welcher Browser wurde verwendet?
War die Seite bereits länger geöffnet?
War die Seite in einem Tab im Hintergrund?
Die meisten Browser pausieren Seiten, welche nicht auf dem aktiven Tab laufen. Das bedeutet auch, dass per JS keine aktuellen Daten empfangen werden.
Wenn das SoC Modul einen neuen Ladestand ermittelt, steht dieser komplett in openWB zur Verfügung. Es macht keinen Unterschied, ob automatisch oder manuell aktualisiert wird.

martin-zh, 2021-02-16:
Tritt auf unter Edge Chromium auf Windows 10 UND Chrome auf Android 10. Seite neu geöffnet.

martin-zh, 2021-02-16:
Ich denke, dass ich es morgen nochmals testen kann (fahre heute nicht wegen Home Office). Soll ich dabei spezielle Daten sammeln?

benderl, 2021-02-16:
Das ist normal nicht erforderlich. Vielleicht noch ein paar Angaben zur openWB:
- gekaufte openWB oder selbst installiert
- welche Softwareversion ist installiert
- welche Hardware ist verbaut (Standard, Standard+, Buchse, Standalone,...)

Im Browser ggf. die JS Konsole öffnen, um eventuelle Fehler zu sehen.

martin-zh, 2021-02-16:
Es ist eine gekaufte Duo, installiert ist 1.9.101 (2021-02-11 12:18:26 +0100 [a086ef87] )

benderl, 2021-03-22:
Tritt das Problem mit der aktuellen Stable 1.9.220 oder neuer noch auf?

martin-zh, 2021-03-22:
Es ist mir seither nicht mehr aufgefallen, ich muss aber sagen dass ich mich auch nicht geachtet habe und aufgrund der Home Office Situation derzeit nicht wirklich viel fahre. 

Ich denke wir können hier schliessen; wenn es nochmals auftreten würde, kann ich ja ein neues Issue aufmachen und auf dieses hier verweisen.

benderl, 2021-03-23:
Danke für die Rückmeldung.

closed 2021-03-23 by benderl