snaptec/openWB #1066: 1.9.101: Smarthome Fehler

martin-zh, 2021-02-16:
Ich erhalte noch folgende Fehlermeldung von SmartHome2:

2021-02-12 10:59:49: Conditions (1) Geschirr Fehlermeldung: '1relais'
2021-02-12 10:59:49: Device tasmota1Geschirr Fehlermeldung: [Errno 2] No such file or directory: '/var/www/html/openWB/ramdisk/smarthome_device_ret1'
KeyError: 'Voltage'
if ( int(answer['StatusSNS']['ENERGY']['Voltage']) > 50 ):
File "/var/www/html/openWB/modules/smarthome/tasmota/watt.py", line 19, in <module>
Traceback (most recent call last):
2021-02-12 10:59:49: Config reRead

Dazu scheint die Leistung nicht ausgelesen zu werden; auf der Übersichtsseite siehts so aus - siehe attachment
[image]

okaegi, 2021-02-16:
Es sieht so aus als OpenWB keine Verbindung zum Tasmota Device aufbauen kann.
a) Bitte mal mit der ipadresse direkt das Tasmota Device im Browser anschauen und dokumentieren.
b) Und die SmarthomE 2.0 definitionen für das Tasmota dokumentieren.
Danke Gruss Oliver

martin-zh, 2021-02-17:
Gerne - siehe attachment. Das Device ist ein Shelly 1PM mit Tasmota drauf. 
[image]
[image]

Wenn ich http://192.168.253.44/cm?cmnd=Status aufrufe, erhalte ich:
{"Status":{"Module":0,"FriendlyName":["Geschirrspueler"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

http://192.168.253.44/cm?cmnd=Power:
{"POWER":"ON"}

http://192.168.253.44/cm?cmnd=Status%208
{"StatusSNS":{"Time":"2021-02-17T13:32:36","Switch1":"ON","ANALOG":{"Temperature":49.0},"ENERGY":{"TotalStartTime":"2021-02-17T00:01:02","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0},"TempUnit":"C"}}

Hilft das weiter?

okaegi, 2021-02-17:
Danke für die detaillierten Unterlagen,
offensichtlich liefert Tasmota auf Shelly nur nach Hardwaremodifikation die Spannung (Voltage) zurück.

{"StatusSNS":{"Time":"2021-02-17T13:32:36","Switch1":"ON","ANALOG":{"Temperature":49.0},"ENERGY":{"TotalStartTime":"2021-02-17T00:01:02","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0},"TempUnit":"C"}}

https://github.com/arendst/Tasmota/pull/9905
Somit schlägt der Zugriff auf  int(answer['StatusSNS']['ENERGY']['Voltage']) fehl. Hier geht es aber nur drum festzustellen ob der Verbraucher läuft. Die aktuelle Leistungsaufnahme Power wird aber auch hier geliefert und diese fragen wir in Zukunft ab.  (> 50, analog http modul). Wird in der nächsten nightly für alle Tasmota  korrigiert

martin-zh, 2021-02-17:
Danke für den schnellen Fix - werde updaten sobald eine neue Nightly da ist. 
Hast du Paypal? Würde dir gerne ein Bier / Kaffee / Glace oder what ever offerieren.

benderl, 2021-02-18:
#1073 sollte damit behoben sein.

closed 2021-02-18 by benderl