snaptec/openWB #1081: Smarthome mit Tasmota: Schaltet nicht ein gemäss GUI und Log

martin-zh, 2021-02-19:
Ist ein followup von https://github.com/snaptec/openWB/issues/1066

Mit der 1.9.103 (2021-02-18 14:10:10 +0100 [48de454b] ) wird nun die Leistung angezeigt. Auch wird der Shelly mit Tasmota eingeschaltet, doch sieht es im Log seltsam aus:

2021-02-19 14:37:56: (1) Geschirr Einschaltverzögerung nicht erreicht. 120 > 5
2021-02-19 14:37:56: (1) Geschirr Ueberschussberechnung (1 = mit Speicher, 2 = mit Offset) 1
2021-02-19 14:37:56: (1) Geschirr SoC 100 Einschalt SoC 0 Ueberschuss 5444
2021-02-19 14:37:56: (1) Geschirr Überschuss 5444 größer Einschaltschwelle1750
2021-02-19 14:37:56: (1) Geschirr Maximale Einschaltdauer nicht erreicht
2021-02-19 14:37:56: Total Watt nichtabschaltbarer smarthomedevices: 0
2021-02-19 14:37:56: Total Watt abschaltbarer smarthomedevices: 0
2021-02-19 14:37:56: (1) Geschirr relais: 0 aktuell: 0 Zaehler Hw: 0
2021-02-19 14:37:56: Speicher Entladung(-)/Ladung(+): 0 SpeicherSoC: 100
2021-02-19 14:37:56: Uberschuss: 5444 Uberschuss mit Offset: 5444
2021-02-19 14:37:56: EVU Bezug(-)/Einspeisung(+): 5444 max Speicherladung: 0
2021-02-19 14:37:51: (1) Geschirr Einschaltverzögerung gestartet
2021-02-19 14:37:51: (1) Geschirr Ueberschussberechnung (1 = mit Speicher, 2 = mit Offset) 1
2021-02-19 14:37:51: (1) Geschirr SoC 100 Einschalt SoC 0 Ueberschuss 5409
2021-02-19 14:37:51: (1) Geschirr Überschuss 5409 größer Einschaltschwelle1750
2021-02-19 14:37:51: (1) Geschirr Maximale Einschaltdauer nicht erreicht
2021-02-19 14:37:51: Total Watt nichtabschaltbarer smarthomedevices: 0
2021-02-19 14:37:51: Total Watt abschaltbarer smarthomedevices: 0
2021-02-19 14:37:51: (1) Geschirr relais: 0 aktuell: 0 Zaehler Hw: 0
2021-02-19 14:37:50: Speicher Entladung(-)/Ladung(+): 0 SpeicherSoC: 100
2021-02-19 14:37:50: Uberschuss: 5409 Uberschuss mit Offset: 5409
2021-02-19 14:37:50: EVU Bezug(-)/Einspeisung(+): 5409 max Speicherladung: 0
2021-02-19 14:37:44: (1) Geschirr angeschaltet. Ueberschussberechnung (1 = mit Speicher, 2 = mit Offset) 1
2021-02-19 14:37:44: (1) Geschirr Einschaltverzögerung erreicht, schalte ein 1750
2021-02-19 14:37:44: (1) Geschirr Ueberschussberechnung (1 = mit Speicher, 2 = mit Offset) 1
2021-02-19 14:37:44: (1) Geschirr SoC 100 Einschalt SoC 0 Ueberschuss 5316
2021-02-19 14:37:44: (1) Geschirr Überschuss 5316 größer Einschaltschwelle1750
2021-02-19 14:37:44: (1) Geschirr Maximale Einschaltdauer nicht erreicht
2021-02-19 14:37:44: Total Watt nichtabschaltbarer smarthomedevices: 0
2021-02-19 14:37:44: Total Watt abschaltbarer smarthomedevices: 0
2021-02-19 14:37:44: (1) Geschirr relais: 0 aktuell: 0 Zaehler Hw: 0
2021-02-19 14:37:44: Speicher Entladung(-)/Ladung(+): 0 SpeicherSoC: 100
2021-02-19 14:37:44: Uberschuss: 5316 Uberschuss mit Offset: 5316
2021-02-19 14:37:44: EVU Bezug(-)/Einspeisung(+): 5316 max Speicherladung: 0
2021-02-19 14:37:39: (1) Geschirr Einschaltverzögerung nicht erreicht. 120 > 120
2021-02-19 14:37:39: (1) Geschirr Ueberschussberechnung (1 = mit Speicher, 2 = mit Offset) 1
2021-02-19 14:37:39: (1) Geschirr SoC 100 Einschalt SoC 0 Ueberschuss 5316
2021-02-19 14:37:39: (1) Geschirr Überschuss 5316 größer Einschaltschwelle1750
2021-02-19 14:37:39: (1) Geschirr Maximale Einschaltdauer nicht erreicht
2021-02-19 14:37:39: Total Watt nichtabschaltbarer smarthomedevices: 0
2021-02-19 14:37:39: Total Watt abschaltbarer smarthomedevices: 0
2021-02-19 14:37:39: (1) Geschirr relais: 0 aktuell: 0 Zaehler Hw: 0
2021-02-19 14:37:38: Speicher Entladung(-)/Ladung(+): 0 SpeicherSoC: 100
2021-02-19 14:37:38: Uberschuss: 5316 Uberschuss mit Offset: 5316
2021-02-19 14:37:38: EVU Bezug(-)/Einspeisung(+): 5316 max Speicherladung: 0
2021-02-19 14:37:33: (1) Geschirr Einschaltverzögerung nicht erreicht. 120 > 114

Kann die openWB den Zustand nicht feststellen und beginnt deshalb gleich wieder eine Einschaltsequenz?  Das verhalten ist das selbe, wenn ich etwas (10W) Last erzeuge. 

Via http://192.168.253.44/cm?cmnd=Status wäre der Status feststellbar:
off:
{"Status":{"Module":0,"FriendlyName":["Geschirrspueler"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
on:
{"Status":{"Module":0,"FriendlyName":["Geschirrspueler"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

okaegi, 2021-02-19:
OpenWb erkennt Tasmota als eingeschaltet wenn die Leistungsaufnahme vom angeschlossenen Verbeaucher > 50 Watt ist. Sofern keine Leistungsaufnahme da ist, schick openWb den Befehl erneut. Bitte mal mit grösserer Last versuchen, 
Gruss Oliver

martin-zh, 2021-02-19:
OK, kann ich testen. Aber wieso wird die Last angeschaut und nicht der Zustand des Schalters?

okaegi, 2021-02-19:
Hallo Martin,
War im Smarthome 1.0 m.e. schon so. Für Shellies mit shelly Firmware wird tatsächlich der Schalterzustand abgefragt, ebenso wie für mystrom Schalter. Ich weiss nicht, ob tasmota für die gebräuchlichen Firmwarestände tatsächlich die Möglichkeit durchgängig  bietet.
Gruss Oliver

martin-zh, 2021-02-23:
Hallo Oliver

Ich hätte an so etwas gedacht:

#!/usr/bin/python3
import sys
import os
import time
import json
import getopt
import socket
import struct
import codecs
import binascii
import urllib.request
named_tuple = time.localtime() # getstruct_time
time_string = time.strftime("%m/%d/%Y, %H:%M:%S tasmota watty.py", named_tuple)
devicenumber=str(sys.argv[1])
ipadr=str(sys.argv[2])
uberschuss=int(sys.argv[3])
answer = json.loads(str(urllib.request.urlopen("http://"+str(ipadr)+"/cm?cmnd=Status%208", timeout=3).read().decode("utf-8")))
aktpower = int(answer['StatusSNS']['ENERGY']['Power'])
answer2 = json.loads(str(urllib.request.urlopen("http://"+str(ipadr)+"/cm?cmnd=Status", timeout=3).read().decode("utf-8")))
r_status = int(answer['Status']['POWER'])
#if ( int(answer['StatusSNS']['ENERGY']['Voltage']) > 50 ):
relais = 0
if r_status = 1:
   relais = 1
if aktpower > 50:
   relais=1
#else:
#   relais=0
powerc = 0
answer = '{"power":' + str(aktpower) + ',"powerc":' + str(powerc) + ',"on":' + str(relais) + '} '
f1 = open('/var/www/html/openWB/ramdisk/smarthome_device_ret' + str(devicenumber), 'w')
json.dump(answer,f1)
f1.close()

So hätte man zusätzlich zum jetzigen Verhalten auch eine Reaktion auf den Status welcher den Zustand des Relais beschreibt. 

Gruss
Martin

tobiforster, 2021-02-23:
Hey zuammen,
evtl. wäre hier ein Fallbackszenario denkbar?
Zuerst den Status der Tasmota-API abfragen, sofern kein vernünftiges Ergebnis zurückkommt, wird der Fallback via > 50Watt angewendet.

Status kann via HTTP-GET `http://10.0.0.5/cm?cmnd=status` abgefragt werden

VG
Tobi

martin-zh, 2021-02-23:
Genau das sollte mein Code machen

tobiforster, 2021-02-23:
zu spät gesehen ;) bzw. gleichzeitig abgeschickt :)
Daumen hoch!

okaegi, 2021-02-23:
Sollte möglich sein , ich baue es demnächst so ein.

martin-zh, 2021-03-02:
Habe gestern auf die Nightly aktualisiert (1.9.111 (2021-03-01 13:30:55 +0100 [c3865825] )). Leider scheint der Relais Status nicht korrekt ausgelesen zu werden. 

2021-03-02 10:01:00: (1) Geschirr Einschaltverzögerung nicht erreicht. 120 > 51
2021-03-02 10:01:00: (1) Geschirr Ueberschussberechnung (1 = mit Speicher, 2 = mit Offset) 1
2021-03-02 10:01:00: (1) Geschirr SoC 100 Einschalt SoC 0 Ueberschuss 2012
2021-03-02 10:01:00: (1) Geschirr Überschuss 2012 größer Einschaltschwelle1750
2021-03-02 10:01:00: (1) Geschirr Maximale Einschaltdauer nicht erreicht
2021-03-02 10:01:00: Total Watt nichtabschaltbarer smarthomedevices: 44
2021-03-02 10:01:00: Total Watt abschaltbarer smarthomedevices: 0
2021-03-02 10:01:00: (1) Geschirr relais: 0 aktuell: 44 Zaehler Hw: 0
2021-03-02 10:01:00: Speicher Entladung(-)/Ladung(+): 0 SpeicherSoC: 100
2021-03-02 10:01:00: Uberschuss: 2012 Uberschuss mit Offset: 2012
2021-03-02 10:01:00: EVU Bezug(-)/Einspeisung(+): 2012 max Speicherladung: 0

martin-zh, 2021-03-02:
Ah - womöglich sind in der Nightly die Änderungen gar nicht mit drin. sorry.

martin-zh, 2021-03-03:
Nun siehts gut aus! Danke!

benderl, 2021-03-03:
Danke für die Rückmeldung.

closed 2021-03-03 by benderl