snaptec/openWB #1108: Verbesserung Powerwall EVU Modul

erikgieseler, 2021-02-28:
Mit dem Powerwall EVU Modul war es bisher nicht möglich Spannung, Stromstärke und Leistung je Phase am Hausanschluss auszulesen,
da die Powerwall API die notwendigen Daten bisher nicht bereitgestellt hatte. 

In der Firmware Version 20.49 werden bei Aufruf der IP https://POWERWALL/api/meters/site die benötigten Werte je Phase als JSON ausgegeben:

Stromstärke [A]:
i_a_current
i_b_current
i_c_current

Spannung [V]:
v_l1n
v_l2n
v_l3n

Leistung [W]:
real_power_a
real_power_b
real_power_c

Anbei die Ausgabe von /api/meters/site meiner Powerwall zur besseren Verdeutlichung:

```
[
   {
      "id":0,
      "location":"site",
      "type":"synchrometerX",
      "cts":[
         true,
         true,
         true,
         false
      ],
      "inverted":[
         false,
         false,
         false,
         false
      ],
      "connection":{
         "short_id":"",
         "device_serial":"",
         "https_conf":{
            
         }
      },
      "Cached_readings":{
         "last_communication_time":"2021-02-28T18:24:32.890776537+01:00",
         "instant_power":5,
         "instant_reactive_power":-52,
         "instant_apparent_power":52.23983154643591,
         "frequency":0,
         "energy_exported":51463.46612202004,
         "energy_imported":873270.02686978,
         "instant_average_voltage":413.33660471823686,
         "instant_total_current":2.0325,
         "i_a_current":1.285,
         "i_b_current":0.6365000000000001,
         "i_c_current":0.14,
         "v_l1n":239.20000000000002,
         "v_l2n":238.09,
         "v_l3n":238.84,
         "last_phase_voltage_communication_time":"2021-02-28T18:24:32.590658642+01:00",
         "real_power_a":-124,
         "real_power_b":119,
         "real_power_c":10,
         "reactive_power_a":64,
         "reactive_power_b":-89,
         "reactive_power_c":-27,
         "last_phase_power_communication_time":"2021-02-28T18:24:32.89073104+01:00",
         "serial_number":"",
         "timeout":1500000000
      }
   }
]
```

Für Rückfragen bzw. Tests stehe ich gerne zur Verfügung.

benderl, 2021-02-28:
Gibt es eine Möglichkeit, die Version der Firmware zu ermitteln?

erikgieseler, 2021-02-28:
Ja, einfach die URL https://POWERWALL/api/status aufrufen

```
{
"start_time": "2021-02-19 14:28:06 +0800",
"up_time_seconds": "221h40m30.939482148s",
"is_new": false,
"version": "20.49.0",
"git_hash": "",
"commission_count": 0,
"device_type": "teg",
"sync_type": "v2"
}
```

benderl, 2021-03-13:
@erikgieseler
Kannst Du einen PR erstellen?

erikgieseler, 2021-03-13:
Ist erstellt: 
https://github.com/snaptec/openWB/pull/1174

closed 2021-03-16 by benderl