snaptec/openWB #1118: Variable heißen kWh, meinen aber Wh

Flohlaus, 2021-03-03:
Guten Abend,

es gibt zahlreiche Variable, die kWh heißen (einspeisungkwh, bezugkwh ...). Auch im Interface z. B. den allgemeinen EVU-Modulen http oder json wird die Eingabe für den Wert "kWh" verlangt und als Muster steht .kWh im Feld.
Der gesuchte Wert wird aber in Wh gerechnet, nicht in kWh.

Grundsätzlich sollte ein Wert nicht nach seiner Einheit benannt sein. Gesucht sind Leistung und Energiemenge, nicht W oder kWh.
Zumindest im User Interface sollte das richtig benannt sein (und lässt sich vermutlich auch schnell umsetzen).

DetMoerk, 2021-03-04:
Ich verstehe den Hintergrund hinter dem Vorschlag, aber überleg mal, wenn jemand jetzt keinen oder wenig technischen Hintergrund hat, dann schaut er auf seinen Zähler oder auf die Webseite seines WR und findet dort eine Angabe kWh und eine Angabe in Watt. Was meinst du ist für diesen User einfacher zuzuordnen? Leistung und Energiemenge oder Watt und kWh.
Wobei man auch sagen muss, dass die spezifischen Module  den Kunden eh weiter an die Hand nehmen und das oft gar nicht verlangen. Bei den universellen http, json... ist oft schon mehr Detailwissen erforderlich.

Nur meine Meinung dazu.

VG
Det

Flohlaus, 2021-03-04:
Das stimmt wohl. Aber wer ein json-Modul manuell nutzt muss mindestens die API vom Wechselrichter lesen. Und dann wäre ich erstmal irritiert, wenn die Eingabe in kWh verlangt wird, der WR aber Wh liefert.
Kann der Laie evtl. mit den Begriffen "Leistung [W]" und "Verbrauch [Wh]" was anfangen?

Marcst76, 2021-03-06:
Schließe mich @Flohlaus an. Auch ich dachte nun erst, Mist, da muss ich noch Dinge umrechnen. Wenn "kWh" nur geschrieben ist, um "Verbrauch in Wh" zu  meinen, ist das ziemlich irritierend.

Flohlaus, 2021-03-08:
Hier mal ein Vorschlag für die Benennungen. Ich denke es ist hilfreich, als Beispiel eine verschachtelte json Antwort zu nehmen, das wird der Standardfall sein, oder nicht?
[image]

benderl, 2021-03-25:
In 1.9.225 behoben.

closed 2021-03-25 by benderl