snaptec/openWB #1119: Logging Daten gekaufte OpenWB

Flohlaus, 2021-03-03:
Guten Abend,

in meiner neu gekauften OpenWB Series2 Standard+ liegen irgendwelche alten Logging Files von 2018, 2019, usw... Die Files sind zwar größtenteils leer, aber das sieht etwas komisch aus.

Hilfreich wäre hingegen in dem Ordner eine Textdatei mit der Kopfzeile der Loggingdaten (Wenn es schon nicht in jede Logdatei direkt geschrieben werden kann). Dann muss man das nicht immer aus dem Forum raussuchen...

[image]

quasel, 2021-03-04:
gerade geguckt iss bei meiner gekauften auch so…

benderl, 2021-03-08:
Da wurde vergessen das Image zu bereinigen. Die alten Logdateien stammen aus einem Entwicklungssystem und können gefahrlos gelöscht werden. Der Fehler wurde beim aktuellen Image, das auf die gekauften Boxen gespielt wird, behoben.

closed 2021-03-08 by benderl