snaptec/openWB #1139: Integration SoC für Skoda

RalfAlbers, 2021-03-07:
Integration eines Moduls zur Abfrage des SoC für Skoda Modelle.
Unter https://pypi.org/project/skodaconnect/ ist schon ein sehr gutes Python-Modul zu finden.

Bucky2k, 2021-03-07:
Ich habe ein neues VAG Modul als Pull request eingestellt - es basiert auch auf EVCC und kann glaube ich auch Skoda. Einfach mal bei Fahrzeugtyp in kleinen buchstaben "skoda" eintragen

#1142

Bucky2k, 2021-03-17:
@RalfAlbers Gern in  der aktuellen Nightly das VAG Modul testen - bisher habe ich nur VW prüfen können

RalfAlbers, 2021-03-17:
ich lade mir morgen mal das Image und teste!

Bucky2k, 2021-03-17:
Brauchst nur auf die aktuelle Nightly updaten

mutsch75, 2021-03-19:
Kann ich bestätigen gleicher Fehler :

2021-03-19 07:04:28: Lp1: Error from evcc-soc: 2021/03/19 07:04:28 invalid vehicle type: skoda

mutsch75, 2021-03-19:
Hab jetzt nicht im Detail nachgesehen aber laut Doku von EVCC werden nur die folgenden Modelle unterstützt "skoda: Skoda (Citygo, Enyaq)" also reine E-FAhrzeuge bei mir handelt es sich aber um einen SuperB iV (Plugin Hybrid)

RalfAlbers, 2021-03-19:
ich habe einen Skoda Citigo-e IV

mutsch75, 2021-03-19:
> ich habe einen Skoda Citigo-e IV

danit ist die Begründung dahin :-)

Bucky2k, 2021-03-19:
Es liegt am nicht aktuellen EVCC Modul in der OpenWB - wir müssen uns noch gedulden

benderl, 2021-04-07:
Mit EVCC (allgemeines Modul für alle implementierten Hersteller kommt in Kürze) und TRONITY stehen zwei Möglichkeiten bereit, den SoC abzufragen.

closed 2021-04-07 by benderl