snaptec/openWB #1143: SOC ZeroNG - eingeschränkte Funktion?

Bucky2k, 2021-03-07:
Wenn ich mich nicht täusche, funktioniert in diesem Modul das verkürzte Intervall nicht. Durch das logische OR wird immer das Standardintervall genutzt (glaube ich):

modules/soc_zerong/main.sh

```
if ( (( $ladeleistung > 800 )) && (( zerotimer < zintervallladen )) ) || (( zerotimer < zintervall )); then
	socDebugLog "Nothing to do yet. Incrementing timer."
	zerotimer=$((zerotimer+1))
	echo $zerotimer > $soctimerfile
else
	echo 0 > $soctimerfile
	socDebugLog "Requesting SoC"
...
```

Bucky2k, 2021-03-08:
Wenn jemand diese Vermutung bestätigt, kann ich das gern korrigieren und ein Update vorschlagen

benderl, 2021-03-09:
Die ersten beiden Vergleiche sind noch in einer Klammer zusammengefasst. Ich sehe da kein Problem.

Bucky2k, 2021-03-09:
Ich hatte den Code für mein VAG/EVCC Modul kopiert und erst darüber die Fehlfunktion erkannt. In Worten besagt die If-Bedingung doch:
Wenn die Ladung aktiv ist und der Soc-Timer kleiner dem Lade-Aktualisierungsintervall ist ODER der Soc-Timer kleiner dem Standby-Aktualisierungsintervall ist, dann "Nothing to do/increment"

Letztere Bedingung ist immer true (da Ladeintdrvall < Standbyintervall) und damit immer vorrangig. Egal was die erste Abfrage ergibt.

Ich wäre echt baff wenn die Abfrage doch ihren Dienst erfüllt, dann habe ich echt einen Knoten im Kopf..

benderl, 2021-03-10:
Sorry, hast Recht! Hatte nicht beachtet, dass dann nur der Timer hochgezählt wird. War gedanklich bei der Bedingung, dass die Abfrage gestartet wird.

Bucky2k, 2021-03-10:
Die Variante im VAG Modul (PR) ist adaptiert vom Teslamodul - die läuft. Soll ich für das ZeroNG einen Bugfix vorschlagen?

benderl, 2021-03-10:
Gerne als PR.

closed 2021-03-11 by benderl