snaptec/openWB #1178: Einstellbares Abfrageintervall für http-SOC-Modul

e-lmo, 2021-03-14:
Die fahrzeugspezifischen SOC-Module haben ein einstellbares Abfrageintervall. Für das http-SOC-Modul habe ich solch eine Möglichkeit bisher nicht gefunden

Bucky2k, 2021-03-18:
Bitte nach Möglichkeit einmal testen:

#1198

Bucky2k, 2021-04-08:
In aktuelle Nightly integriert - bitte testen!

e-lmo, 2021-04-08:
Danke, habe es gerade getestet. 
Auf der Hauptseite wird der SOC seit dem Neustart mit nightly mit 0% angezeigt. 
Unter EV_SOC_log finden sich erfolreiche SOC-Werte:
`2021-04-08 15:35:26: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:35:07: Lp1: SoC: 65 (LV0)
2021-04-08 15:35:06: Lp1: Requesting SoC (LV0)
2021-04-08 15:34:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:34:29: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:34:06: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:33:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:33:26: Lp1: Nothing to do yet. Incrementing timer.
2021-04-08 15:33:06: Lp1: Nothing to do yet. Incrementing timer.
2021-04-08 15:32:49: Lp1: SoC: 65 (LV0)
2021-04-08 15:32:49: Lp1: Requesting SoC (LV0)
2021-04-08 15:32:27: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:32:06: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:31:48: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:31:26: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:31:06: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:30:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:30:29: Lp1: SoC: 60 (LV0)
2021-04-08 15:30:28: Lp1: Requesting SoC (LV0)
2021-04-08 15:30:07: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:29:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:29:26: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:29:06: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:28:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:28:26: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:28:07: Lp1: SoC: 60 (LV0)
2021-04-08 15:28:06: Lp1: Requesting SoC (LV0)
2021-04-08 15:27:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:27:27: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:27:07: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:26:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:26:25: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:26:06: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:25:47: Lp1: SoC: 60 (LV0)
2021-04-08 15:25:46: Lp1: Requesting SoC (LV0)
2021-04-08 15:25:26: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:25:06: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:24:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:24:26: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:24:06: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:23:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:23:26: Lp1: SoC: 60 (LV0)
2021-04-08 15:23:26: Lp1: Requesting SoC (LV0)
2021-04-08 15:23:06: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:22:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:22:26: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:22:06: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:21:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:21:27: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:21:07: Lp1: SoC: 60 (LV0)
2021-04-08 15:21:06: Lp1: Requesting SoC (LV0)
2021-04-08 15:20:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:20:26: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:20:06: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:19:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:19:26: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:19:06: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:18:46: Lp1: SoC: 60 (LV0)
2021-04-08 15:18:46: Lp1: Requesting SoC (LV0)
2021-04-08 15:18:26: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:18:06: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:17:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:17:26: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:17:06: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:16:46: Lp1: Charging, but nothing to do yet. Incrementing timer. (LV0)
2021-04-08 15:16:26: Lp1: SoC: 60 (LV0)
2021-04-08 15:16:26: Lp1: Requesting SoC (LV0)
[... 904 lines cut ...]
2021-04-08 10:31:26: Lp1: Nothing to do yet. Incrementing timer. (LV0)
2021-04-08 10:31:06: Lp1: Nothing to do yet. Incrementing timer. (LV0)
2021-04-08 10:30:46: Lp1: Nothing to do yet. Incrementing timer. (LV0)
2021-04-08 10:30:26: Lp1: Nothing to do yet. Incrementing timer. (LV0)
2021-04-08 10:30:06: Lp1: Nothing to do yet. Incrementing timer. (LV0)
[... cut ...]
2021-04-08 10:26:07: Lp1: Nothing to do yet. Incrementing timer. (LV0)
2021-04-08 10:25:46: Lp1: Nothing to do yet. Incrementing timer. (LV0)
2021-04-08 10:25:29: Lp1: Error from http call (LV0)
2021-04-08 10:25:26: Lp1: Requesting SoC (LV0)
2021-04-08 10:25:06: Lp1: Nothing to do yet. Incrementing timer. (LV0)
2021-04-08 10:24:46: Lp1: Nothing to do yet. Incrementing timer. (LV0)
2021-04-08 10:24:26: Lp1: Nothing to do yet. Incrementing timer. (LV0)
2021-04-08 10:24:09: Lp1: Error from http call (LV0)
2021-04-08 10:24:06: Lp1: Requesting SoC (LV0)`

benderl, 2021-04-08:
Bitte nochmal ein Update machen. Es war ein Tippfehler drin.

e-lmo, 2021-04-09:
Es funktioniert, Danke!

closed 2021-04-09 by e-lmo