snaptec/openWB #1213: Automatischer wechsel von PV auf PVmin abhängig von SOC und PV Ertragserwatung

mutsch75, 2021-03-20:
Hab das bei mir schon ausserhalb von openWB implementiert und steuere die Umschaltung via openwb/api.

Aktulle hole ich mir die Globalstrahlung für meinen Standort für die nächsten 4 Tage, und errechne damit und meiner peak Leistung der PV Analage die Ertragserwartung. 
Wenn nicht zu erwaten ist dass heute oder morgen ein bestimmter Wert zu erreichen ist und SOC in meinem Fall <40%, schalte ich von PV (default) auf PV min um und ggf auch zurück.

Sofern nicht PVmin oder PV aktiv ist wird nichts verändert.

Flohlaus, 2021-03-23:
Ist da nicht auch eine Kombination aus PV und Zielladen sinnvoll?
Wenn abends (oder von mir aus morgen abend) nach Sonnenuntergang SOC unter 40%, dann Sofortladen. Das bringt dann auch eine höhere Ladeeffizienz als Min+PV.

mutsch75, 2021-03-23:
Ok ich hätte vielleicht sagen sollen, das Min+PV bei mir auf 16A steht. Sofortladen nutze ich quasi als "manuelles Laden"

benderl, 2021-03-24:
Ich verstehe das hier nicht ganz.
Soll das ein Wunsch nach einer Erweiterung in openWB sein? Dann muss ich Dich enttäuschen. Lademodi werden in absehbarer Zeit nicht automatisch gewechselt. Wenn das per externer Logik funktioniert, ok.

closed 2021-03-26 by mutsch75