snaptec/openWB #1254: IGMPv3 für SMA Home Manager Abfrage

Becker884, 2021-03-30:
Hallo,
könnten Sie die Abfrage des SMA HM IGMPv3 kompatibel gestalten.
Sobald man unter 
-Modulkonfiguration EVU
--Strombezugsmessmodul (EVU-Übergabepunkt) 
den Home Manager auswählt und dieser abgefragt wird, wird das gesamte Netzwerk von IGMPv3 auf IGMPv2 runter gestuft.

Danke

benderl, 2021-04-07:
Ich habe im Code keine Stelle gefunden, die IGMPv2 erzwingt. Leider sind meine Netzwerkkenntnisse zum Thema Multicast überschaubar.

Becker884, 2021-04-07:
Danke für die Rückmeldung. Ich habe leider noch weniger Kenntnisse davon.

Ich nutze z.B. zusätzlich noch eine Node-Red Abfrage über UDP-in:
[image]
um den SMA HM abzufragen (von einem anderen RPi3 aus) und das stört überhaupt nicht den IGMPv3.

MartinRinas, 2021-04-08:
Der Unterschied zwischen v2 und v3 ist wohl marginal, v3 unterstützt lediglich die Angabe der Quelladresse. Was wohl in großen Umgebungen relevant sein kann, für unseren Fall hier in einem einzelnen, geswitchten Netzwerk jedoch keine Rolle spielen dürfte.
https://ipwithease.com/comparison-of-igmpv1-igmpv2-and-igmpv3/

Becker884, 2021-06-08:
Hallo, ich wollte noch mal anmerken, dass es zu Aussetzern kommt durch die Nutzung des SMA Home Manager EVU Moduls der openWB. Konkret kommt es fast täglich zu Aussetzern der Zählerkommunikation SBS2.5 <-> SHM, entsprechend wird dann Netzstrom bezogen für 30-60s. 
Das ganze tritt auch beim Kollegen auf, der auch eine SBS2.5 und openWB verwendet.
Ich habe ein paar Wochen openWB ausgeschaltet und nicht einen Fehler gehabt.
Habe mir erst mal zur Not über das http EVU Modul was gebastelt, schön ist das aber nicht.

Becker884, 2021-06-25:
Eine mögliche einfache Lösung wäre es, auf die Abfrage des Energy Meters (SHM) **ganz zu verzichten**, und sich die EVU Werte per **Modbusabfrage** des SBS 2.5 (oder neueren SB/STP) zu holen.