snaptec/openWB #1256: Kann ich die Sonnenbatterie 10 integrieren

miwi-1209, 2021-04-01:


DetMoerk, 2021-04-01:
Ich denke die Frage ist im Forum besser aufgehoben als hier.

https://openwb.de/forum/index.php

Aber am besten erst mal mir einer Suche beginnen im Forum. 

VG
Det

benderl, 2021-04-07:
Hat sich gegenüber der Version 8 die API geändert?
Ruf mal im Browser folgende URL auf: `<IP-der-Sonnenbatterie>/api/v1/status`

ginan2, 2021-04-08:
Hallo,

bei mir funktioniert es als Sonnen Eco Batterie:

In den EVU Einstellungen:
[image]

Unter Batteriespeicher Einstellung:
[image]

LG

ginan2

benderl, 2021-04-12:
Damit sollte die Frage beantwortet sein.

closed 2021-04-12 by benderl