snaptec/openWB #1299: Sonderzeichen Passwort ( VAG, ID, Audi, EVCC Modul)

Bucky2k, 2021-04-11:
In den EVCC basierten Modulen wird die EVCC-binary wie folgt aufgerufen:

answer=$($MODULEDIR/../evcc-soc-abo $fztype --user "$username" --password "$password" --vin "$vin" --token "$token" 2>&1)

Durch die Shell-Interpretation werden Sonderzeichen aber interpretiert unter z.B. --password "$password". Ein einfaches Ersetzen der " gegen ' (bei allen Variablen) hat ad-hoc nicht funktioniert. Kann mir wer sagen, wie das Problem bzw. die Einschränkung umgangen werden kann? Z.B. verlangt Porsche ein Sonderzeichen im Passwort und die SoC Abfrage funktioniert daher nicht bei allen (mutmaßlich in Abhängigkeit des gewählten Sonderzeichens).

benderl, 2021-04-12:
Zumindest bei dem etwas älteren Python argparse Modul gibt es einen Bug. Um Sonderzeichen komplett an den Prozess zu übergeben, muss die Syntax `... --param='value'` benutzt werden. Wie das in Go aussieht, kann ich nicht beurteilen. Da kann aber sicher AndiG etwas zu sagen. Singlequotes sind bei Kennwörtern zwingend erforderlich.

andig, 2021-04-12:
> Zumindest bei dem etwas älteren Python argparse Modul gibt es einen Bug. Um Sonderzeichen komplett an den Prozess zu übergeben, muss die Syntax `... --param='value'` benutzt werden. Wie das in Go [...]

Parsen tut m.e. die Shell und nicht Go. Dort wird nur noch argv übergeben.

Bucky2k, 2021-04-12:
Wie gesagt, ich kenne generell die Syntax hier nicht, ich habe nur aus anderen Modulen übernommen. Ändere ich:

`answer=$($MODULEDIR/../evcc-soc-abo $fztype --user "$username" --password "$password" --vin "$vin" --token "$token" 2>&1)`

in

`answer=$($MODULEDIR/../evcc-soc-abo $fztype --user "$username" --password '$password' --vin "$vin" --token "$token" 2>&1)`

bekomme ich Fehler aus dem EVCC Modul zurück. Ich belasse es erstmal bei den double-quotes, schlicht weil mir der Ansatz fehlt wie das robuster gemacht werden kann. Bucky

benderl, 2021-04-13:
Eventuell wird es mit #1306 behoben.

benderl, 2021-04-13:
> Wie gesagt, ich kenne generell die Syntax hier nicht, ich habe nur aus anderen Modulen übernommen. Ändere ich:
>
> `answer=$($MODULEDIR/../evcc-soc-abo $fztype --user "$username" --password "$password" [...]

Mit den Singlequotes unterbindest Du jegliche Variablenexpansion. Dem evcc wird dann buchstäblich $password übergeben.

benderl, 2021-04-20:
Nach dem PR #1325 sollte es eigentlich funktionieren. Bitte testen.

benderl, 2021-04-24:
Wie sieht es aus? Abgesehen von ' sollte es mit allen Sonderzeichen funktionieren. Könnt ihr das bestätigen?

Bucky2k, 2021-04-24:
Müsste es schon einfach so funktionieren oder muss im Modul im Aufruf noch
von " auf ' beim PW gewechselt werden?

benderl, 2021-04-24:
Nein, ansonsten übergibst Du wieder $password an evcc, ohne Variablenexpansion.

Bucky2k, 2021-04-24:
Soeben ein Passwort mit Ziffern, Groß/Kleinschreibung und #, ! und & angelegt und erfolgreich den SOC abrufen können. Ich würde sagen läuft

benderl, 2021-04-24:
Dann schließe ich mal ab.

closed 2021-04-24 by benderl