snaptec/openWB #1300: Aufruf der updateConfig.sh nicht immer zuverlässig?

Bucky2k, 2021-04-11:
Hi,

ich hatte selbst gestern das Problem während der EVCC-Modulerstellung, dass für manche Anpassungen die openWB.conf betreffend erst ein Reboot notwendig war, um eine korrekt ergänzte openWB.conf zu erhalten. Auch im Forum, z.B. bei der Nutzung des VAG Moduls, hatte ich Kontakt zu zwei Usern und die Abhilfe war am Ende ein Downgrade auf Stable, Update zurück auf Nightly und ein abschließender Reboot. 

Kann es sein, dass aktuell nach einem Update die updateConfig.sh oder die loadconfig.sh nicht sauber aufgerufen werden? Vielleicht bin ich auch auf der falschen Fährte und es war Zufall. Aber es war nun schon verdächtig, dass der Rücksprung auf Stable und/oder der Reboot bei mehreren Usern notwendig waren und Abhilfe brachte.

Gruß

Bucky

benderl, 2021-04-12:
Das kann nur anhand von Logdateien geprüft werden. `updateConfig.sh` schreibt in die `openwb.log` wann es startet und wieder fertig ist. Während des Updates / nach einem Reboot wird die Regelschleife gezielt abgebrochen, um die Initialisierung der `openwb.conf` und Ramdisk abzuwarten.

Bucky2k, 2021-04-12:
Ich werde erstmal weiter beobachten - evtl. bleibt es ja bei den Einzelfällen

Bucky2k, 2021-04-26:
Ich konnte erstmal nichts weiteres feststellen, ggf. ein temporäres Problem zwischen einem Versionsschritt. Ich schließe hier erstmal

closed 2021-04-26 by Bucky2k