snaptec/openWB #1344: error in apache LogFile

eisenkoch, 2021-04-28:
im Logfile des Apache2 habe ich folgende Meldungen gefunden
>>>>>>>>>>>>>>>>
`[Wed Apr 28 09:41:45.120611 2021] [php7:warn] [pid 27886] [client 192.168.178.61:38174] PHP Warning:  file_get_contents(/var/www/html/openWB/ramdisk/ladestarts2): failed to open stream: No such file or directory in /var/www/html/openWB/web/api.php on line 102
[Wed Apr 28 09:41:55.132895 2021] [php7:warn] [pid 419] [client 192.168.178.61:38204] PHP Warning:  file_get_contents(/var/www/html/openWB/ramdisk/ladestart): failed to open stream: No such file or directory in /var/www/html/openWB/web/api.php on line 100
[Wed Apr 28 09:41:55.133116 2021] [php7:warn] [pid 419] [client 192.168.178.61:38204] PHP Warning:  file_get_contents(/var/www/html/openWB/ramdisk/ladestarts1): failed to open stream: No such file or directory in /var/www/html/openWB/web/api.php on line 101
`
ist das so i.o oder sollten wir das iinitRamdisk.sh anpassen?

benderl, 2021-04-28:
Gerne die fehlenden Dateien ergänzen.

benderl, 2021-04-28:
Danke

closed 2021-04-28 by benderl