snaptec/openWB #1411: Tag releases

IngmarStein, 2021-05-27:
Currently, builds are released by bumping the version in https://github.com/snaptec/openWB/blob/master/web/version. Releases are designated as beta or stable by merging them into the respective branches.

This makes it hard to build a given version from source. For example, to build 1.9.220, one would need to find the commit which bumped the version number to that string and build from that commit. This makes it infeasible to provide Docker images corresponding to specific OpenWB releases.

Would you consider tagging releases with git tags?

benderl, 2021-05-27:
We will switch to regular releases with version 2.0.

ChrisMuki, 2022-01-16:
Using releases or tags instead of that version file would also make the whole project feel more mature 😉

closed 2022-07-17 by IngmarStein