snaptec/openWB #1417: Nutzung der neuen Go-E Changer Funktion via api V2

dataeasy74, 2021-06-01:
HAllo es wäre nett wenn die neuen GoE Charger Funktionen in die openWB SW implementiert werden könnten Danke und Grüße

snaptec, 2021-06-01:
Wir sind mit Go-e in Klärung wie eine weitere Zusammenarbeit aussieht.

sebnaf, 2021-07-07:
Hallo,

da ich die Möglichkeit habe, meinen go-e von HW v2 auf v3 zu aktualisieren:
Funktionieren die heutigen Funktionen mit der v3?

Danke

andig, 2021-07-07:
Was kann Go-E v3 denn Neues?

sebnaf, 2021-07-07:
Verbesserte Hardware, Cloudanbindung… - das Beste aber: Phasenumschaltung (1/3) per API und sogar während dem Laden.

alex15977, 2021-07-14:
> Verbesserte Hardware, Cloudanbindung… - das Beste aber: Phasenumschaltung (1/3) per API und sogar während dem Laden.

Ja, man kann in der App einstellen, dass noch die API v1 verwendet wird, dann funktioniert OpenWB wie zuvor.
PS: Für die API v2 gibt es seitens go-e auch noch keine Spezifikation, man könnte hier also noch gar nichts entwickeln (bis auf openWB selbst, die evtl. an mehr Infos kommen...)

etraider, 2021-09-20:
Die Doku ist seit dem 14.9. draußen: 
https://github.com/goecharger/go-eCharger-API-v2/blob/main/introduction-de.md

Interessant wäre evtl. die Variable "psm" (1 Phase="1", 3 Phasen = "2"), die ist aber leider nicht dokumentiert.

matzempc, 2021-11-13:
Auch die "alte" API (Version 1) hat einen neuen Parameter erhalten, der zur Phasenumschaltung dient (fsp) dieser steht bei 3Phasen auf 0 und bei 1Phase auf 1.
Dementsprechend habe ich die 1-/3-Phasenumschaltung als Python Skript implementiert und es an die Phasenumschaltungsfunktionalität der openWB angebunden. Das Skript erkennt automatisch ob eine Phasenumschaltung möglich ist (und es sich somit um HW Version 3 handelt) am Vorhandensein des Parameters (dies konnte ich allerdings nicht wirklich testen, da ich "nur" einen Go-e mit HW Version 3 habe aber keinen "alten"). 
Die Umschaltung klappt bei mir wie gewünscht (mein Setup openWB series2 standard+ und als zweiten LP einen Go-e).
Die Änderungen können auf folgendem branch meines forks eingesehen werden:
https://github.com/matzempc/openWB/tree/goe-u1p3p-switch
Hier wird die APIv1 für die Umschaltung verwendet.
Ein zweiter Branch beinhaltet die Nutzung der neuen APIv2 in der gesamten openWB Software (und somit auch Umstellung des Python Skripts zur Phasenumschaltung):
https://github.com/matzempc/openWB/tree/goe_api_v2

matzempc, 2021-11-13:
Gerne würde ich das als Pull Request einstellen, wenn es berechtigte Hoffnung gibt, dass dieser auch gemerged werden würde.

matzempc, 2021-11-13:
Falls sich jemand fragt warum ich im 1-/3-Phasenumschaltungshandling der openWB (u1p3pcheck.sh) die Änderung der Stromwerte deaktiviert habe: Sollte ein HW Version 2 Go-e als Ladepunkt konfiguriert sein, könnte es hier negative Seiteneffekte geben. Zudem wie oben geschrieben managt zum einen der Go-e bei Umschaltung (fast) alles selbst und ich habe in meinem Python Skript ein Setzen der StromWertes auf den MinPV Wert bereits berücksichtigt und implementiert.

alex15977, 2021-11-13:
> Auch die "alte" API (Version 1) hat einen neuen Parameter erhalten, der zur Phasenumschaltung dient (fsp) dieser steht bei 3Phasen auf 0 und bei 1Phase auf 1.
> Dementsprechend habe ich die 1-/3-Phasenumschaltung als Python Skript implementiert und es an die Phasenumschaltungsfunktionalität der openWB angebunden. Das Skript erkennt automatisch ob eine Phasenumschaltung möglich ist (und es sich somit um HW Version 3 handelt) am Vorhandensein des Parameters (dies konnte ich allerdings nicht wirklich testen, da ich "nur" einen Go-e mit HW Version 3 habe aber keinen "alten").
> Die Umschaltung klappt bei mir wie gewünscht (mein Setup openWB series2 standard+ und als zweiten LP einen Go-e) mit meinem Tesla Model 3.
> Der Go-e übernimmt das gesamte Management der Umschaltung bei umsetzen des Parameters völlig selbständig (Stoppen,umschalten, anstarten).
> Die Änderungen können auf folgendem branch meines forks eingesehen werden:
> https://github.com/matzempc/openWB/tree/goe-u1p3p-switch
> Hier wird die APIv1 für die Umschaltung verwendet.
> Ein zweiter Branch beinhaltet die Nutzung der neuen APIv2 in der gesamten openWB Software (und somit auch Umstellung des Python Skripts zur Phasenumschaltung):
> https://github.com/matzempc/openWB/tree/goe_api_v2

Vielen lieben Dank! Vorallem jetzt zur kalten Jahreszeit sehr sinnvoll. 
Gleich mal morgen testen :)

andig, 2021-11-14:
> Auch die "alte" API (Version 1) hat einen neuen Parameter erhalten, der zur Phasenumschaltung dient (fsp) dieser steht bei 3Phasen auf 0 und bei 1Phase auf 1.

Das würde implizieren, dass auch die alte HW die Fähigkeit hat? Schwer vorstellbar?

matzempc, 2021-11-14:
Nein, ich gehe natürlich nicht davon aus, dass die alte HW diese Fähigkeit hat...wie sollte sie auch, dass ist tatsächlich eine HW-Eigenschaft, die nicht einfach (sw-technisch) nachgerüstet werden kann.
Ich nehme an, dass der Parameter fsp (in APIv1) bei alter HW entwder gar nicht vorhanden oder mit "null" besetzt ist?! Hier wäre mir auch sehr willkommen, dass jemand mit HW V2 Auskunft darüber erteilen kann.
Fakt ist aber, dass für HW V3 auch die "alte" APIv1 diesen neuen Parameter erhalten hat und darüber die Phasenumschaltung funktioniert - und das sehr gut!

xerion3800, 2021-11-14:
Ich habe diesen Thread durch Zufall entdeckt. Ich habe einen go-e charger V2 mit FW 0.41. Der Parameter "fsp" taucht dort gar nicht auf.

sebnaf, 2021-11-14:
> Ich habe diesen Thread durch Zufall entdeckt. Ich habe einen go-e charger V2 mit FW 0.41. Der Parameter "fsp" taucht dort gar nicht auf. 

Du verwendest einen sehr alten und instabilen Softwarestand. Bitte aktualisiere Deine Firmware.

xerion3800, 2021-11-14:
> > Ich habe diesen Thread durch Zufall entdeckt. Ich habe einen go-e charger V2 mit FW 0.41. Der Parameter "fsp" taucht dort gar nicht auf.
> 
> Du verwendest einen sehr alten und instabilen Softwarestand. Bitte aktualisiere Deine Firmware.

Mein letzter Wissenstand ist, das die 0.41 das aktuellste für die V2 ist. Die App zeigt auch keiner neue SW an. Bei V3 gibt es eine 0.5x aber nicht beim V2

sebnaf, 2021-11-14:
Ah - sorry. Ich war gedanklich bei der v3. 
Ich glaube die v2 kann "fsp" noch gar nicht.

matzempc, 2021-11-15:
> Ich habe diesen Thread durch Zufall entdeckt. Ich habe einen go-e charger V2 mit FW 0.41. Der Parameter "fsp" taucht dort gar nicht auf.
So hatte ich das auch erwartet! Entsprechend wird dann vom JSON Parser NULL zurückgegeben.
Leider ist mir aber aufgefallen, dass beim apiv2 branch hier noch etwas fehlt, gefixt werden muss, mach ich heute abend!
Danach sollte aber automatisch APIv1 bzw. APIv2 erkannt und benutzt werden.
Der Phasenumschaltungsbranch sollte jedoch für HW3 funktionieren und bei HW2 keinen Einfluss haben.

benderl, 2022-07-27:
Die Phasenumschaltung wird von openWB offiziell nur bei eigener Hardware unterstützt. Da seitens Go-e keine Zusammenarbeit gewünscht ist, wird es hierfür keine Anpassungen geben.
Eine Unterstützung der neueren API müsste von jemandem entwickelt werden, der entsprechende Hardware zum Testen hat. PR (ohne Phasenumschaltung!) darf gerne gestellt werden.
Daher schließe ich hier ab.

closed 2022-07-27 by benderl

andig, 2022-07-27:
Umschaltung und v2 Api gibts bei evcc. Eine Zusammenarbeit mit Go-e haben wir leider auch nicht :(

benderl, 2022-07-27:
Wenn Du das in Eigenregie verantwortest und für eventuelle Schäden haftest, kann EVCC alles machen. Seitens openWB gab es bereits genügend Diskussion hierzu im Forum. Weitere Diskussionen hier sind nicht erwünscht.

andig, 2022-07-27:
Ich sehe hier hier keine Diskussion zu openWB?

sebnaf, 2022-07-27:
> Eine Zusammenarbeit mit Go-e haben wir leider auch nicht :(

Ich hatte mehrfach angeboten zu vermitteln…