snaptec/openWB #1438: Support dynamic evcc vehicles selection

andig, 2021-06-16:
See 

```
❯ curl https://cloud.evcc.io/api/vehicles
[{"id":"audi","name":"Audi"},{"id":"bmw","name":"BMW"},{"id":"citroen","name":"Citroën"},{"id":"default","name":"Default"},{"id":"ford","name":"Ford"},{"id":"hyundai","name":"Hyundai"},{"id":"kia","name":"Kia"},{"id":"nissan","name":"Nissan"},{"id":"carwings","name":"Nissan (Leaf pre 2019)"},{"id":"opel","name":"Opel"},{"id":"ovms","name":"Ovms"},{"id":"peugeot","name":"Peugeot"},{"id":"porsche","name":"Porsche"},{"id":"renault","name":"Renault"},{"id":"seat","name":"Seat"},{"id":"vw","name":"VW"},{"id":"id","name":"VW ID"},{"id":"volvo","name":"Volvo"},{"id":"skoda","name":"Škoda"},{"id":"enyaq","name":"Škoda Enyaq"}]
```

benderl, 2021-06-17:
@Bucky2k 
Willst Du das noch umsetzen?

Bucky2k, 2021-06-17:
Grundsätzlich ja, aber noch ist ja kein Bedarf entstanden. Die Statische Liste ist meines Wissens nach noch vollständig. Vielleicht habe ich nächste Woche etwas Luft und Langeweile

andig, 2021-06-17:
Ich glaube aktuell fehlt noch Carwings in der Auswahl für die alten Leaf. Ist aber sicher kein Massenphänomen.

andig, 2021-07-21:
EVCC hat in master jetzt auch Unterstützung für den Fiat500e. Per Cloud könnte ich das sofort verfügbar machen- würde aber davon abhängen, die Liste zu erweitern oder zu dynamisieren. Könntet ihr Euch das nochmal anschauen? Vielen Dank!!!

Bucky2k, 2021-07-22:
Mangels Zeit erstmal nur das Dropdown erweitert: #1496

@andig: "fiat" als Übergabestring für den Fahrzeugtyp ist korrekt?

andig, 2021-07-23:
Super, danke 🙏

andig, 2021-07-23:
Ich schaue mal ob ich die dynamische Auswahl per copy/paste hinbekomme. Nur testen kann ichs mangels owb leider nicht.

Bucky2k, 2021-07-23:
Sende mir den vorbereiteten Codeschnipsel gern zu - testen bekomme ich bestimmt noch unter

andig, 2021-07-24:
Ich hab mir das angeschaut, aber blicke nicht durch. Die jquery Hölle in einer 5000 Zeilen langen Datei ist nicht meins. Muss also fürs Erste bleiben wie es ist :(

ts78, 2021-08-06:
Ich habe die API Abfrage eingefügt, getestet und als Pull Request gestellt.

andig, 2021-08-06:
Top, vielen Dank @ts78 !!!

ts78, 2021-08-06:
@andig: Für den Nutzer wäre es im Übrigen übersichtlicher, wenn Škoda nach Seat und vor VW kommt - hast du Einfluss auf die Reihenfolge der Items?

andig, 2021-08-06:
Nur wenn ich die Liste con Hand baue. Es scheint kein entsprechendes Unicode Sort zu geben in Go :(

andig, 2021-08-06:
Nvm. Natürlich gibts das. Ändere ich asap…

andig, 2021-08-07:
Ist drin, danke für den Hinweis. Ich mag mal wieder wie einfach das in Go ist:

	collator := collate.New(language.Make("de"))

	sort.Slice(types, func(i, j int) bool {
		return collator.CompareString(types[i].Name, types[j].Name) < 0
	})

Einfach aus `strings.Compare` ein `collator.Compare` gemacht.

PR #1511 (2021-08-07): Settings: evcc: Support dynamic vehicles selection
Use public API from evcc cloud to get available/supported vehicles.
This fixes #1438
commit 4e524ceb (2021-08-07): Settings: evcc: Support dynamic vehicles selection

Use public API from evcc cloud to get available/supported vehicles.
This fixes #1438
closed 2021-08-07 by benderl