snaptec/openWB #1458: PIN-Eingabefeld "0" fehlt

zzTAz, 2021-06-24:
Wenn unter /openWB/web/settings/misc.php

[image]

Das Theme auf Symbolfluss gestellt wird und das Display durch eine PIN-Abfrage geschützt ist,
besteht das Problem, dass beim Entsperren des Displays das Zahlenfeld lediglich die Zahlen 1-9 als Eingabe zulässt. 
Die 0 sucht man hier vergebens.

dergreg, 2023-06-16:
@benderl laut blame ist das gauges CSS von dir, das lässt sich doch sicherlich recht schnell fixen, oder? (falls es denn allein am CSS liegt) 🙂

Wollte letztens auch meine OpenWB entsperren und hab die 0 vergeblich gesucht.

benderl, 2023-06-17:
Weder das Theme noch der CSS Teil sind von mir.
Unser Fokus liegt jetzt bei der 2.0 und da kann das Theme so nicht verwendet werden.