snaptec/openWB #1498: MQTT Default setting "try_private true" is causing problems with regular MQTT brokers

one4many, 2021-07-27:
https://github.com/snaptec/openWB/blob/948c5a1c35b6fe08549255c1c346154124e1962b/web/settings/savemqtt.php#L311

The try_private default setting in the MQTT config section of OpenWB is causing major problems for NON-Mosquitto brokers which do not support this setting. This setting causes the setting of 132 (hex 0x84) in the protocol version field of the MQTT header, which in turn makes it incompatible to other MQTT brokers (violation of the protocol). A non mosquitto broker expects here the proper setting of 4 (hex 0x04) for MQTT version 3.1.1 for example.

In other words setting try_private to true is for mosquitto brokers/bridges only and pretty much incompatible with the rest of the world. It should be set to false here to actually make MQTT useful to the other systems without hassling the OpenWB user to change this setting by hand (I mean edit config files via SSH / every time they change anything in the MQTT settings, because the 'local' change gets overwritten, suse-yast style every time).

Please set:
`try_private false`
as a default setting

dj3mu, 2021-08-12:
I agree with the arguments. I didn't take care of this in the initial implementation.

So I fully agree that `try_private false` is the more universal configuration.

But now I'm a bit concerned about compatiblity for existing users of the MQTT bridge feature (once they re-configure their bridge) as we don't know and cannot test all setups.

I'm testing against a Mosquitto v 2.0.11 broker now and, so far, didn't see any issue setting `try_private false`.

The change itself is very obviously very easy.
@snaptec do you want me to create PR for it? Or do you have any additional concerns?

benderl, 2021-08-13:
I would suggest to make it configureable. For the openWB cloud it should be set to true. Any other manual bridge condiguration should default to false for compatibility.

dj3mu, 2021-08-13:
> 
> 
> I would suggest to make it configureable. For the openWB cloud it should be set to true. Any other manual bridge condiguration should default to false for compatibility.

Working on it. PR will follow once finished and tested.

dj3mu, 2021-08-20:
The feature has been merged into latest (Nightly) openWB version.
@one4many Please verify if it fits your needs and close this issue or comment if you still have problems.

closed 2021-09-05 by benderl

one4many, 2021-09-09:
@dj3mu Works for me now. Thank you!