snaptec/openWB #1500: screenshots4wiki

m4ding, 2021-07-28:
man kann sonst nirgendswo bilder im wiki hoch laden f[image]
[image]
[image]
r Artikel...

m4ding, 2021-07-28:
[image]

JSAnyone, 2021-07-28:
Du musst das Issue nicht abschicken. Reicht wenn du die Bilder hochlädst, dann den Link von den Bildern kopierst und das Issue verwirfst.

m4ding, 2021-08-01:
sehr gut, danke!

closed 2021-08-04 by m4ding