snaptec/openWB #1577: Falscher Preis bei preisbasierendem Laden

nottheedge, 2021-09-19:
Hallo,

leider wird beim preisbasierenden Laden der falsche Strompreis von Awattar angezeigt. Es fehlt die Mehrwertsteuer. Sollte diese nicht mit einberechnet werden? 

[image]
[image]

Ich verwende die OpenWB Series 2 custom mit Version: 1.9.245

Viele Grüße und schönen Sonntag,
Jörg

riesi77, 2021-09-27:
Woher kommt der Screenshot mit den Preisen? Ich vergleiche immer mit https://www.awattar.at/services/charts/hourly und da stimmen die Preise überein. Ich denke, das sind die Netto Preise - was auch OK ist, weil es unterschiedliche Mehrwertsteuersätze in den Ländern gibt.

nottheedge, 2021-09-27:
Der Screenshot kommt von meiner OpenWB. Richtig ist, dass die MwSt unterschiedlich sein kann, aber im Code ist der Faktor 1,19 für DE hinterlegt... Somit müsste doch dann der richtige Bruttopreis angezeigt werden?

[image]

riesi77, 2021-09-29:
Der untere Screenshot kommt auch von der OpenWB?

nottheedge, 2021-09-29:
Der untere Screenshot ist von einer App, welche die Preise von Awattar abholt. Das diese Preise mit denen von Awattar übereinstimmen, zeigen folgende aktuelle Screenshots: 
[image]
[image]
[image]

riesi77, 2021-09-29:
Genau der Name dieser App würde mich interessieren - weil sie hilfreich scheint.

nottheedge, 2021-09-29:
Kein Problem. Here we go: [AWattPrice](https://apps.apple.com/de/app/awattprice/id1536629626)

MiOrt, 2021-10-06:
Die Diskussion um die Anzeige der aWATTAr-Preise wurde in der Vergangenheit schon mehrfach geführt. Per Definition ist ein Fehler die "Nichterfüllung einer Anforderung" (EN ISO 9000:2005). Es handelt sich bei dem beschriebenen Phänomen also nicht um einen Fehler, denn die Software arbeitet exakt so, wie es gefordert ist. Es ist eher eine Abweichung von einer persönlichen Präferenz.

Da auch der Begriff "Strompreis" nicht definiert ist, kommen wir auch mit der Anzeige des API-Preises + MwSt nicht weiter, denn dann kommt die nächste Beschwerde (zurecht), dass der Preis nicht dem Bruttoarbeitspreis entspricht, der höchstwahrscheinlich in anderen Apps angezeigt wird. Egal wie man es dreht... die Steuerung und korrekte Funktion der openWB ist mit der aktuell gewählten Anzeigeart vollständig gegeben. Man könnte vielleicht den Begriff "Strompreis" im GUI in "Nettostrompreis" umbenennen, aber das würde einen erheblichen Mehraufwand bedeuten (denn Tibber liefert per API den Bruttoarbeitspreis und dieser wird im gleichen Feld angezeigt). Ansonsten ist ja bereits alles vorbereitet, um - sobald man den Basispreis in der Konfiguration eingeben kann - den korrekten Bruttoarbeitspreis anzuzeigen. Aber auch hier... dann stimmt dieser Preis ebenfalls nicht mit Deiner App überein, da dort anscheinend eben nur Börse + MwSt berechnet wird (was ja auch nicht korrekt ist, denn als Kunde zahlt man eben noch einiges mehr).

Selber machen kann das natürlich jeder, auch gerne die Änderungen zur Speicherung und Verarbeitung des Basispreises vornehmen. Empfehlen würde ich das aber **zum derzeitigen Zeitpunkt** nicht, da die Umstellung auf 2.0 bereits im Gange ist und solche Änderungen natürlich auch dort implementiert sein sollten/müssen. Über die Nightly Updates laufen alle Änderungen, die per Pull Request in Github durch die Besitzer des Repos übernommen werden. Dann sind sie für alle per Update verfügbar. Vielleicht sagt Lutz hier ja noch was dazu.