snaptec/openWB #1595: OS Update needed for BMW SoC to keep on working

dj3mu, 2021-10-01:
As mentioned already in PR https://github.com/snaptec/openWB/pull/1594 BMW is using a LetsEncrypt certificate for its server(s) hosting `https://www.bmw-connecteddrive.com/api/vehicle`.

Due to expiry of cross-signing certificate, an `apt update && apt upgrade -y` was necessary on my openWB to get trust anchors for the new LetsEncrypt root CA !

How will this be handled for openWBs without root access? I don't think any update UI feature is available for OS updates yet.

I think this will be an issue for many "older" openWBs.

benderl, 2021-10-01:
Updating installed packages couold be done in `atreboot.sh`. We already learned the hard way how to spawn a new process which does not get killed if the parent one dies. So this should work in normal booting state and after triggering an update.

To prevent not needed updates, I would suggest to set a flag in the update settings to do this once. Otherwise package updates are triggered on every boot.

snaptec, 2021-10-01:
Shouldn't downloading the root CA like mentioned here
https://www.stephenwagner.com/2021/09/30/sophos-dst-root-ca-x3-expiration-problems-fix/
fix this?

DetMoerk, 2021-10-01:
Das Problem ist wohl eher, dass die openWB beim API Zugriff auf ein Problem stößt, wenn das entsprechende Zertifikat nicht in der Chain ist.
Ich würde auf das Paket openssl tippen, was aktualisiert werden muss.

DetMoerk, 2021-10-01:
Kleiner Test von meinem Raspi4 unter einem recht aktuelle Buster
`pi@raspi4:~/xxx $ curl https://www.bmw-connecteddrive.com/api/vehicle
pi@raspi4:~/xxx $`

Jetzt die openwb mit dem Stretch aus eurer Auslieferung

`pi@openWB:~/xxx $ curl https://www.bmw-connecteddrive.com/api/vehicle
curl: (60) SSL certificate problem: certificate has expired
More details here: https://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.
pi@openWB:~/xxx $
`

DetMoerk, 2021-10-01:
Vermutlich reicht auch das Paket ca-certificates

DetMoerk, 2021-10-02:
https://openwb.de/forum/viewtopic.php?f=12&t=4112

Vermute das Thema breitet sich jetzt aus.

VG
Det

dj3mu, 2021-10-02:
@DetMoerk Vielen Dank für die Untersuchungen welche Pakete nötig sind. Dazu hatte ich gestern keine Zeit. 

Mein Vorschlag wäre in `atreboot.sh` dann das folgende Kommando so hinzuzufügen wie von @benderl in https://github.com/snaptec/openWB/issues/1595#issuecomment-931975324 beschrieben (Ausführung im Hintergrund):
```
apt update && apt install -y ca-certificates openssl curl
```
Möglicherweise ginge noch weniger wenn man explizit `dpkg-reconfigure ca-certificates` aufruft.  
Aber einmal ausgeführt wird es aufwändig zu testen da explizites Downgrade von Paketen eine eher komplizierte Angelegenheit ist. Wenn also obiges funktioniert sollten wir es dabei lassen. Das Risiko, dass da noch was unerswünschtes mit kommt dürfte sehr gering sein.

Nebenbei: Das komplette "apt upgrade" hat bei mir auch keinerlei Probleme verursacht. Aber ich halte es dennoch für besser, nur das zu aktualisieren, was wirklich nötig ist um die neuen Root-Zertifikate zu bekommen.

snaptec, 2021-10-02:
Ist in der nightly gefixt:
https://github.com/snaptec/openWB/commit/a05d6b98616996dd1a436c548e5882d9ef9f6089

closed 2021-10-02 by snaptec