snaptec/openWB #1600: Slave: Anzeige der Zählerdaten falsch

s0170071, 2021-10-01:
Ich hab zwei neue OpenWBs , eine mit Display (Master, Ladepunkt 1 (Garage) ) und eine ohne (slave, Ladepunkt 2 (Einfahrt) ).
Kernel: Linux 4.19.66-v7+ GNU/Linux
openWB Version: 1.9.220 

Wenn ich beim Master die Übersicht anzeige, werden die Slave (LP2) Spannungsdaten in der Übersicht nicht angezeigt.

[image]

Wenn ich aber den Slave ebenfalls als Standalone Box betreibe, werden die Werte (jetzt als LP1) korrekt angezeigt: 
[image]

DetMoerk, 2021-10-01:
Am Besten eine Support Meldung aus der Box aufmachen. 
Einstellungen/System/Support

s0170071, 2021-10-01:
Habs nochmal konfiguriert. Ladepunkte -> externe openWB. Ich glaub ich hatte da Series1/2 Duo stehen. Das Duo - ist mir erst jetzt aufgefallen, ist ja eine eigene Version der Box.

s0170071, 2021-10-01:
Jetzt hab ich allerdings an einem Ladepunkt den Power Factor stehen, am anderen nicht. Feature ?

[image]