snaptec/openWB #1670: Negative Werte... Neue spur

JanFellner, 2021-10-30:
Neue Spur beim Thema negative Werte....
Leider hat es auf mein letztes Ticket zum Thema wenig Bewegung seitens der Analyse gegeben.
Daher neuer Versuch mit neuer Erkenntnis.

Gestern habe ich ~21kwh in meinen ioniq geladen. Heute schaue ich gerade rein nachdem ich ne push Benachrichtigung bekommen dass gerade mit dem Laden begonnen wurde.

Was sehen meine müden Augen da. - 21kwh in der status Übersicht.

Kann der Wert beim Tageswechsel negativ übernommen worden sein? 
[image]
[image]

Beme99, 2022-02-13:
Hallo Jan, ich habe das gleiche Problem, ebenfalls mit dem Modul SimpleEVSEWifi. Ich nutze einen S0 Zähler in der Wallbox und überlege ob es daran liegen könnte. Was hast Du für einen Zähler? Auch S0 oder Modbus?
Viele Grüße, Benjamin

Beme99, 2022-02-27:
Vergesst meinen vorherigen Eintrag, bei mir ist openWB nicht die Ursache.
Ich nutze einen S0-Zähler, und EVSE-Wifi liefert für den Zählerstand (meterReading in getParameters) rückläufige Werte, d.h. der Zählerstand steigt nicht, sondern fällt beim neu anstecken. 
Habe das Problem bereits bei EVSE-Wifi gemeldet: https://board.evse-wifi.de/viewtopic.php?f=4&t=341&p=2383#p2383

JanFellner, 2022-03-18:
Hi @Beme99,
haben die den Forumsbereich gelöscht? Ich kann da nix aufrufen obwohl ich dort nen Account habe...

https://board.evse-wifi.de/viewtopic.php?f=4&t=341&p=2383#p2383
-> nicht berechtigt

Oder hast du nen aktuellen Status?
Jan

Beme99, 2022-03-20:
Hallo Jan, 
das ist ein Bereich mit speziellen Zugriffsberechtigungen (Beta-Tester).
Mathias, der Entwickler der EVSE-Wifi hat sich am 04.03. im Thread gemeldet:

> [](https://board.evse-wifi.de/viewtopic.php?f=4&t=341&p=2383#top)Danke fürs Melden, dieses Bugs. Da hab ich was im Kopf, dass wir dieses Problem eigentlich schon mal geslöst hatten. Aber ich werde mir das nochmal genauer ansehen, wie das zustande kommt. Ich denke, das werden wir lösen können.

Viele Grüße
Benjamin

benderl, 2022-03-20:
Da dies anscheinend kein Fehler bei openWB ist, schließe ich hier mal ab.

closed 2022-03-20 by benderl