snaptec/openWB #1704: Mindest-SOC Einstellung verschwindet nach Aktivieren von mehr als einem Ladepunkt

Brovning, 2021-11-14:
Hallo,

wenn ich nur 1 Ladepunkt konfiguriere, kann ich den Maximal-Soc und den Mindest-Soc unter "Ladeeinstellungen" --> "PV- und Min-PV" konfigurieren:

[image]

Sobald ich 2 Ladepunkte konfiguriere verschwindet die Möglichkeit einen Mindest-Soc zu konfigurieren und man kann nur noch einen Maximal-Soc konfigurieren:

[image]

Ich gehe davon aus, dass es sich hierbei um einen Bug handelt.

installierte Version: 1.9.244

benderl, 2021-11-14:
Das ist kein Bug, sondern "works as designed". Eine Implementierung würde einen größeren Umbau der Regelung nach sich ziehen.

Brovning, 2021-11-14:
Oh, das ist aber ungünstig.
Wie kann ich eine Mindestladung mit 2 Wallboxen realisieren?

benderl, 2021-11-14:
Derzeit ist es nicht automatisiert möglich. Mit der Version 2.0 (derzeit in Entwicklung) ist das auch mit mehreren Ladepunkten machbar.

Brovning, 2021-11-14:
Was ist der geplante Veröffentlichungszeitpunkt der v2.0?

benderl, 2021-11-15:
Wenn es keine größeren Probleme mehr gibt, wird noch dieses Jahr eine erste Testversion veröffentlicht. Die ist dann jedoch mit Vorsicht zu genießen und eher für freiwillige Tester und Entwickler gedacht.

closed 2021-11-15 by benderl