snaptec/openWB #1750: Colors Theme - gewählter Preisbereich wird nicht markiert

riesi77, 2021-11-29:
Wird im Colors Theme (Web) preisbasiertes Laden aktiviert und ein Wert gewählt, der innerhalb des Preisbereichs fällt, wird dieser in anderen Themes markiert, sodass man weiß, es wird für x Stunden geladen werden.
Im Colors Theme ist das leider nicht der Fall. Keine Markierung. Damit weiß man leider auch nicht genau, wie lange geladen werden würde.

closed 2021-11-29 by benderl

riesi77, 2021-12-01:
@benderl : Sollte das in der aktuellen Nightly schon funktionieren? Ich sehe leider noch keine Markierung:
[image]

benderl, 2021-12-01:
@cshagen kannst Du das bitte prüfen?

cshagen, 2021-12-01:
Ist ein Bug. Muss ich fixen.

cshagen, 2021-12-02:
Gefixt im PR  #1762

riesi77, 2021-12-03:
@cshagen , @benderl : Gerade in der nightly überprüft: Funktioniert perfekt. Herzlichen Dank!