snaptec/openWB #1757: Mqtt Topics für Nur Ladepunkt Modus inkonsistent

andig, 2021-11-30:
Aus dem Code:

```
openWB/set/isss/ClearRfid
openWB/set/isss/Cpulp1
openWB/set/isss/Current
openWB/set/isss/heartbeat
openWB/set/isss/Lp2Current
openWB/set/isss/parentCPlp1
openWB/set/isss/parentCPlp2
openWB/set/isss/parentWB
openWB/set/isss/U1p3p
openWB/set/isss/U1p3pLp2
```

Inkonsistenzen: 
- lp vorne vs hinten 
- lp groß vs klein
- lp2 angehängt, lp1 nicht
- Groß vs. Kleinschreibung des Topics

Wäre super, dass in v2 aufzuräumen

/cc @snaptec

andig, 2022-01-21:
Mir fehlt auch das Topic um dem Ladepunkt eine RFID mitzugeben?

andig, 2022-07-07:
Um den Ladepunkt programmatisch freischalten zu können, z.B. wenn ein bestimmtes Fahrzeug verbunden ist. Automatische Fahrzeugerkennung gibts bei evcc.

benderl, 2022-07-07:
Evcc sendet also einen Tag an den Ladepunkt, damit der diesen Tag an Evcc zur Freischaltung sendet? Interessanter Ansatz, aber irgendwie seltsam.

hhoefling, 2022-07-07:
git's irgendwo eweitere Infos...
Außer GPS fällt mir da nix ein.
Und wie gesagt ob auch der Stecker drin steckt weiss du ja erst wenn Strom beim Auto ankommt. (messbar)
Wenn ja dann war es wohl das richtige Auto
Aber zur Freigabe taugt das nicht..

benderl, 2022-07-07:
> Was ist daran seltsam? Wenn meine Wallbox öffentlich erreichbar ist möchte ich, dass mein Fahrzeug mit PV lädt, aber nicht beliebige Fahrzeuge der Nachbarn. Natürlich kann ich mein RFID Tag auch dran halten, aber das geht auch automatisch. Oder ginge falls OpenWB die tag injection ermöglichen würde wie das z.B. bei der Warp der Fall ist.

Seltsam finde ich den Umweg über den Ladepunkt. Eine openWB als externer Ladepunkt hat keine eigene Sperre, sondern wird komplett über den Master gesteuert.

hhoefling, 2022-07-07:
Oh, ich denke EVCC will in diesem Fall den Master spielen.

benderl, 2022-10-20:
@andig  Sind die Tests abgeschlossen?

andig, 2022-10-20:
Jetzt hast Du mich auf dem falschen Fuß erwischt- welche Tests? Das Thema hier war ja konsistentes Naming und dazu gabs bisher kein Update?

benderl, 2022-10-20:
War auf #2294 bezogen. In 1.9 werden die Namen nicht mehr angepasst.

andig, 2022-10-20:
Hab leider kein Feedback erhalten. Die Testerin war nach den end- und sinnlosen Runden über den Support wohl erschöpft. Machen wir hier zu als wontfix bis 2.0?

benderl, 2022-10-20:
???

closed 2022-10-20 by benderl

andig, 2022-10-20:
> ???

Es gab x roundtrips mit dem Support zur CP Unterbrechung bei Nur Ladepunkt. Am Ende war die Erkenntnis aus #2294, dass das heute einfach nicht umgesetzt ist/funktioniert. Wenn Du da mehr Details willst lass uns gerne per Slack reden.

benderl, 2022-10-20:
Ich bin da nicht der richtige Ansprechpartner.

andig, 2023-11-01:
@benderl gibts denn für 2.0 eine neue Mqtt Struktur? Welche?

benderl, 2023-11-06:
In 2.x ist vorgesehen, dass die Anbindung von Drittanbieter-Software über Modbus/TCP erfolgt.