snaptec/openWB #1761: Nur Ladepunkt Modus: konfigurierten Strom auf Mqtt ausgeben

andig, 2021-12-01:
Im Nur Ladepunkt Modus wäre schön wenn
- [ ] `openWB/lp/<num>/AConfigured` den über `openWB/set/isss/Current` eingestellten Ladestrom zeigen würde und
- [ ] `openWB/lp/<num>/ChargePointEnabled` entsprechend Strom >0 `0/1` anzeigen würde

hhoefling, 2021-12-06:
Wenn ich es richtig verstehe dann "leben" die Variablen "openWB/lp/<num>/AConfigured"
nur im der Master-OpenWB. Daraus wird dann der Befehlt zur Slave abgeleitet.
Also schau mit MQTT des Masters nach.

andig, 2021-12-06:
> Also schau mit MQTT des Masters nach.

Das wird bei einem Standalone Slave wie z.B. der Pro schwierig.

hhoefling, 2021-12-06:
Wenn kein Master da ist, dann ist der "Nur Ladepunkt" modus nicht der richtige.

andig, 2021-12-06:
Der Master kann ohne weiteres eine andere Anwendung sein, z.B. https://evcc.io. Und genau dann möchte ich sehen wie der Slave gerade parametriert ist.

andig, 2022-01-08:
Spricht was dagegen den aktuellen Stand einfach mit auszugeben?

andig, 2022-08-17:
Ping

andig, 2022-08-17:
> Wenn kein Master da ist, dann ist der "Nur Ladepunkt" modus nicht der richtige.

@hhoefling brauchst Du mehr Erläuterung?

> Spricht was dagegen den aktuellen Stand einfach mit auszugeben?

benderl, 2022-08-17:
Sorry, aber das hat bei OpenWB derzeit absolut keine Priorität. Da hilft auch kein Ping.

andig, 2022-08-17:
Allerdings hilft eine Aussage ;)