snaptec/openWB #1776: Daten werden nicht korrekt dargestellt Plenticore vs. OpenWB

Krachbummente78, 2021-12-06:
Moin Moin 

Es sieht so aus als sind Netzbezug und Hausverbrauch vertauscht. Die Daten sind ok aber das Labelling schein nicht korrekt zu sein. 

[image]

Die Daten stammen Kostal Plenticore. 
[image]

Gruß aus Flensburg

MiOrt, 2021-12-07:
SW Version? Log-Einträge? Wo sitzt der Energymanager, welche Module wie konfiguriert?

Grundsätzlich stimmen die Anzeigen aus dem WR und der openWB nicht überein, da Kostal einige Werte anders interpretiert. Das hatten wir schön öfter auch im Forum dargestellt. Aber... in den letzten 2 Jahren habe ich so einen "Bug" nicht feststellen können. Die openWB fragt alle 10s den oder die WR ab und berechnet daraus den Hausverbrauch. Dieser ist eine rein rechnerische Größe, da üblicherweise der Energymanager am Hausanschlusspunkt sitzt. Aufgrund zeitlicher Unterschiede der gelesenen Werte, ist dann natürlich auch der angezeigte Wert ein anderer. 

Wie kommst Du darauf, dass Netz und Haus vertauscht sind? Die Werte unterscheiden sich in den Anzeigen doch schon erheblich, selbst wenn man sie tauscht?

Krachbummente78, 2022-01-12:
[image]
[image]
Die Werte werden in meinen Augen falsch angezeigt bzw. missinterpretiert. Das Haus verbraucht ca -300Watt und im WB Display steht ich würde die Menge aus dem Energienetz beziehen und Hausverbrauch ist aktuelle ca 5 Watt. Ich denke da liegt das Problem. Ist das einigermaßen verständlich? Ich ziehe kein Strom aus dem Netz sondern aus der PV aber die WB geht davon aus ich würde es aus dem Netz beziehen.

MiOrt, 2022-01-12:
And again... welche Softwareversion, wo sitzt der Energymanager, welche Module wie konfiguriert?? 

Deine Interpretation der Anzeige ist falsch. Zuerst... die Anzeige im Wechselrichter und der Box laufen nicht synchron, weil die Darstellungs- und Anzeigeintervalle nicht überein stimmen! Und dann zu Deiner Anzeige... Laut Box erzeugst Du 373W, lädst 36W in den Speicher speist 332W ins Netz und das Haus braucht 5W. Hausverbrauch ist nur ein berechneter Wert aus den anderen 3 und der Ladeleistung. Da die Datensätze nicht immer zeitlich übereinander passen (sequenzielles aller Register etc.), stimmt 1. die Anzeige nicht mit dem WR überein und 2. kommt es auch mal zu unplausiblen Anzeigen.

benderl, 2022-01-21:
So als Laie, was den Plenticore betrifft: eventuell die Einstellung der Einbauposition falsch gesetzt?