snaptec/openWB #1792: EVCC Zugriffe

hhoefling, 2021-12-10:
Beim Laden von modulconfiglp.php werden diese Zeilen ausgeführt.

Zeile 48:
			// read available vehicles for EVCC SoC module from web API
			$soc_evcc_vehicles = json_decode(file_get_contents('https://cloud.evcc.io/api/vehicles'), true);
			$firstentry=array("id" => "none", "name" => "Bitte auswählen");
			array_unshift($soc_evcc_vehicles , $firstentry);

Es werden die dort verfügbaren PKW eingeliste um für den Fall das der User das EVCC Module auswählen würde
eine PKW Liste anzeigen zu können.
Da ich kein EVCC Module verwend denke ich das es schöner wäre wenn das Zugriff auf EVCC erst erfolgen würde wenn das EVCC Module auch wirklich verwendet wird. 

Falls der dortige Server rumzickt, dann ziggt openWB auch und läuft beim Seitenaufbau auf einen Timeout.
Sehr unschön wenn mann EVCC gar nicht auf dem Schirm hat.

benderl, 2021-12-15:
@andig Kannst Du da bitte mal nachsehen? Wäre sicher gut, wenn die Liste erst geladen wird, wenn sie auch benötigt wird, heißt im Browser mit JS. Als Beispiel kann das Tronity Modul genutzt werden.

andig, 2021-12-15:
Da kann uns nur @Bucky2k helfen- ich habe keine OpenWB in der ich implementieren kann...

Bucky2k, 2021-12-23:
@benderl Hi, die API abfrage hattest Du doch glaube ich selbst eingebaut oder wer war das? Derzeit wird der Abruf in Zeile 48 in der Tat für alle openWB Nutzer ausgeführt - das ist ja auch für EVCC und deren Server unnötige Last. Ich bin mir aber nicht sicher, wie dieser Request für die Listen-API direkt in das EVCC Modul verschoben werden kann/muss, damit es funktioniert.

Gruß

Bucky

benderl, 2022-01-18:
Nein, das habe ich nicht "verbrochen". ;-)
https://github.com/benderl/openWB/commit/4e524ceb34b14316e2d10ca3d09d4ad338db801d

Ich hatte eben einen kurzen Test analog zu Tronity gemacht. Leider wurde das Abrufen der Liste wegen aktiviertem CORS auf dem EVCC Server vom Browser abgebrochen. Die unterstützten Fahrzeugtypen müssten dann von einem PHP-Skript auf der openWB abgerufen werden, welches wiederum in der modulconfiglp.php per Ajax abgerufen wird.

Freiwillige vor.

andig, 2022-01-18:
> Leider wurde das Abrufen der Liste wegen aktiviertem CORS auf dem EVCC Server vom Browser abgebrochen. 

Das verstehe ich nicht. Abrufen der Liste geht doch bzw. ging bisher. In welchem Fall geht das jetzt hier nicht? Was sagt die Browserkonsole?

andig, 2022-01-18:
@benderl CORS ergänzt:

```
< HTTP/2 200
< access-control-allow-headers: Accept, Content-Type, Content-Length, Accept-Encoding, X-CSRF-Token, Authorization
< access-control-allow-methods: GET, OPTIONS
< access-control-allow-origin: *
< content-type: application/json; charset=UTF-8
< content-length: 781
< date: Tue, 18 Jan 2022 15:36:25 GMT
```

benderl, 2022-01-18:
Die Browser sind da restriktiver als PHP. Wird etwas von einer anderen Domain per JS geladen, dann wird geprüft, ob der Server dieser Domain das erlaubt. Dazu gibt es den CORS Header. Wenn es komplett im Browser laufen soll, müssten die Header angepasst werden.

Alternativ wird eine Anfrage an die openWB gestellt, welche dann in PHP von EVCC die Liste abruft und weitergibt.

andig, 2022-01-18:
>  Wenn es komplett im Browser laufen soll, müssten die Header angepasst werden.

Done- haben wir uns überschnitten.

benderl, 2022-01-19:
Bitte mal #1947 testen.

closed 2022-01-20 by benderl