snaptec/openWB #1827: Fehler im HHTP Skript

xerion3800, 2021-12-17:
Habe im aktuellen Nightly folgende Logeinträge. Bei mir stellt es sich so da das meine Speicher und EVU Abfrage per HTTP nicht mehr funktioniert:
`AttributeError: 'NoneType' object has no attribute 'group'
    device_config["configuration"]["protocol"] = regex.group(1)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 81, in read_legacy
    read_legacy(sys.argv)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 132, in <module>
Traceback (most recent call last):
2021-12-17 13:56:26: PID: 30936: root: Fehler im HTTP Skript`

yankee42, 2021-12-17:
Wie hast du das Modul konfiguriert?

xerion3800, 2021-12-17:
Am Beispiel von EVU so:
[image]

LKuemmel, 2021-12-17:
Die Korrektur aus PR #1825 wurde gerade mit PR #1829 gemergt. Bitte nochmal probieren.

yankee42, 2021-12-17:
> Die Korrektur aus PR #1825 wurde gerade mit PR #1829 gemergt. Bitte nochmal probieren.

Der Fehler tritt an einer anderen Stelle auf, dass wurde durch 1829 nicht gelöst. Fix incoming.

matzempc, 2021-12-17:
Fuer hinterlegte URLs auf Phase1 bis Phase3 funktioniert es jetzt wieder. Wenn man allerdings "none" hinterlegt (so wie in der Beschreibung, wenn man diese Werte nicht zur Verfügung hat vorgesehen):
[image]

Gibt es weiterhin python exceptions da leider das none nicht korrekt verarbeitet wird:
AttributeError: 'PreparedRequest' object has no attribute 'host'
    return "Die Verbindung zum Server {} ist fehlgeschlagen. Überprüfe Adresse und Netzwerk.".format(e.request.host)
  File "/var/www/html/openWB/packages/helpermodules/exceptions/requests.py", line 8, in handle_connection_error
    result = entry.handler(exception)
  File "/var/www/html/openWB/packages/helpermodules/exceptions/registry.py", line 31, in translate_exception
    return exceptions.get_default_exception_registry().translate_exception(exception)
  File "/var/www/html/openWB/packages/modules/common/fault_state.py", line 75, in from_exception
    fault_state = FaultState.from_exception(exception)
  File "/var/www/html/openWB/packages/modules/common/component_context.py", line 20, in __exit__
    self._components[component].update()
  File "/var/www/html/openWB/packages/modules/http/device.py", line 63, in update
    dev.update()
  File "/var/www/html/openWB/packages/modules/http/device.py", line 122, in read_legacy
    read_legacy(sys.argv)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 131, in <module>
Traceback (most recent call last):

During handling of the above exception, another exception occurred:

requests.exceptions.ConnectionError: HTTPConnectionPool(host='192.168.2.13none', port=80): Max retries exceeded with url: / (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnection object at 0x75c28870>: Failed to establish a new connection: [Errno -2] Name or service not known',))
    raise ConnectionError(e, request=request)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 487, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 609, in send
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 488, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 56, in request
    return request('get', url, params=params, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 70, in get
    response = requests.get(url, timeout=5)
  File "/var/www/html/openWB/packages/modules/http/api.py", line 10, in request_value
    powers = [request_value(self.domain + config["power_l"+str(i)+"_path"]) for i in range(1, 4)]
  File "/var/www/html/openWB/packages/modules/http/counter.py", line 40, in <listcomp>
    powers = [request_value(self.domain + config["power_l"+str(i)+"_path"]) for i in range(1, 4)]
  File "/var/www/html/openWB/packages/modules/http/counter.py", line 40, in update
    self._components[component].update()
  File "/var/www/html/openWB/packages/modules/http/device.py", line 63, in update
Traceback (most recent call last):

During handling of the above exception, another exception occurred:

requests.packages.urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='192.168.2.13none', port=80): Max retries exceeded with url: / (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnection object at 0x75c28870>: Failed to establish a new connection: [Errno -2] Name or service not known',))
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 363, in increment
    _stacktrace=sys.exc_info()[2])
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 643, in urlopen
    timeout=timeout
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 423, in send
Traceback (most recent call last):

During handling of the above exception, another exception occurred:

requests.packages.urllib3.exceptions.NewConnectionError: <requests.packages.urllib3.connection.HTTPConnection object at 0x75c28870>: Failed to establish a new connection: [Errno -2] Name or service not known
    self, "Failed to establish a new connection: %s" % e)
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 147, in _new_conn
    conn = self._new_conn()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 163, in connect
    self.connect()
  File "/usr/lib/python3.5/http/client.py", line 892, in send
    self.send(msg)
  File "/usr/lib/python3.5/http/client.py", line 949, in _send_output
    self._send_output(message_body)
  File "/usr/lib/python3.5/http/client.py", line 1157, in endheaders
    self.endheaders(body)
  File "/usr/lib/python3.5/http/client.py", line 1206, in _send_request
    self._send_request(method, url, body, headers)
  File "/usr/lib/python3.5/http/client.py", line 1161, in request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 361, in _make_request
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 594, in urlopen
Traceback (most recent call last):

During handling of the above exception, another exception occurred:

socket.gaierror: [Errno -2] Name or service not known
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
  File "/usr/lib/python3.5/socket.py", line 733, in getaddrinfo
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 75, in create_connection
    (self.host, self.port), self.timeout, **extra_kw)
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 138, in _new_conn
Traceback (most recent call last):
2021-12-17 17:49:52: PID: 3153: root: Fehler im HTTP Skript

yankee42, 2021-12-17:
> Fuer hinterlegte URLs auf Phase1 bis Phase3 funktioniert es jetzt wieder.

Nein, daran hat sich noch nichts geändert. Bei dir funktioniert es weil du eine IP-Adresse eingegeben hast. Bei xerion3800 stört sich das Script am "localhost".

> Wenn man allerdings "none" hinterlegt (so wie in der Beschreibung, wenn man diese Werte nicht zur Verfügung hat vorgesehen):

Da hast du allerdings recht. Potentiell gefixt durch #1832. Getestet ist das aber nicht.

yankee42, 2021-12-20:
@xerion3800 , @matzempc theoretisch sind die fixes gemergt, gerne nochmal mit dem aktuellen nightly testen.

matzempc, 2021-12-20:
Habe es getestet, war leider noch ein kleiner bug im Zugriff auf das Dictionary bei counter.py drin, habe ich kurz gefixt => passt jetzt (außerdem habe ich den umbau nun auch verstanden ;-)
PR: #1842

xerion3800, 2021-12-20:
Leider keine Verbesserung. Habe nach wie vor keine Werte für Netz, Haus und Speicher und folgende Einträge im Debug Log:
`KeyError: 'power_path'
    self.__get_power = create_request_function(domain, component_config["power_path"])
  File "/var/www/html/openWB/packages/modules/http/bat.py", line 26, in __init__
    component_config, domain)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 52, in add_component
    device.add_component(component_config)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 94, in run_device_legacy
    run_device_legacy(create_legacy_device_config(power_path), component_config)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 117, in read_legacy_bat
    parser.set_defaults(RUN=lambda args: function(*[getattr(args, argument_name) for argument_name in arg_spec.args]))
  File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 19, in <lambda>
    args.RUN(args)
  File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 33, in run_using_positional_cli_args
    {"bat": read_legacy_bat, "counter": read_legacy_counter, "inverter": read_legacy_inverter}
  File "/var/www/html/openWB/packages/modules/http/device.py", line 146, in <module>
Traceback (most recent call last):
KeyError: 'power_all_path'
    self.__get_power_all = create_request_function(domain, component_config["power_all_path"])
  File "/var/www/html/openWB/packages/modules/http/counter.py", line 28, in __init__
    component_config, domain)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 52, in add_component
    device.add_component(component_config)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 94, in run_device_legacy
    run_device_legacy(create_legacy_device_config(power_all_path), component_config)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 131, in read_legacy_counter
    parser.set_defaults(RUN=lambda args: function(*[getattr(args, argument_name) for argument_name in arg_spec.args]))
  File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 19, in <lambda>
    args.RUN(args)
  File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 33, in run_using_positional_cli_args
    {"bat": read_legacy_bat, "counter": read_legacy_counter, "inverter": read_legacy_inverter}
  File "/var/www/html/openWB/packages/modules/http/device.py", line 146, in <module>`

> @xerion3800 , @matzempc die fixes sind gemergt, gerne nochmal mit dem aktuellen nightly testen.

Leider keine Verbesserung. Habe nach wie vor keine Werte für Netz, Haus und Speicher und folgende Einträge im Debug Log:
`KeyError: 'power_path'
    self.__get_power = create_request_function(domain, component_config["power_path"])
  File "/var/www/html/openWB/packages/modules/http/bat.py", line 26, in __init__
    component_config, domain)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 52, in add_component
    device.add_component(component_config)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 94, in run_device_legacy
    run_device_legacy(create_legacy_device_config(power_path), component_config)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 117, in read_legacy_bat
    parser.set_defaults(RUN=lambda args: function(*[getattr(args, argument_name) for argument_name in arg_spec.args]))
  File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 19, in <lambda>
    args.RUN(args)
  File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 33, in run_using_positional_cli_args
    {"bat": read_legacy_bat, "counter": read_legacy_counter, "inverter": read_legacy_inverter}
  File "/var/www/html/openWB/packages/modules/http/device.py", line 146, in <module>
Traceback (most recent call last):
KeyError: 'power_all_path'
    self.__get_power_all = create_request_function(domain, component_config["power_all_path"])
  File "/var/www/html/openWB/packages/modules/http/counter.py", line 28, in __init__
    component_config, domain)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 52, in add_component
    device.add_component(component_config)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 94, in run_device_legacy
    run_device_legacy(create_legacy_device_config(power_all_path), component_config)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 131, in read_legacy_counter
    parser.set_defaults(RUN=lambda args: function(*[getattr(args, argument_name) for argument_name in arg_spec.args]))
  File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 19, in <lambda>
    args.RUN(args)
  File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 33, in run_using_positional_cli_args
    {"bat": read_legacy_bat, "counter": read_legacy_counter, "inverter": read_legacy_inverter}
  File "/var/www/html/openWB/packages/modules/http/device.py", line 146, in <module>`

xerion3800, 2021-12-20:
Oh sehe gerade  das es wohl mit PR: #1844 gelöst sein soll. Dann warten wir mal aufs mergen ;-)

xerion3800, 2021-12-20:
Habe den PR #1844 gerade mal manuell getestet @matzempc . Also Netz und Haus sieht wieder gut aus aber beim Speicher gibt es noch KeyError:
`KeyError: 'power_path'
    self.__get_power = create_request_function(domain, component_config["power_path"])
  File "/var/www/html/openWB/packages/modules/http/bat.py", line 26, in __init__
    component_config, domain)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 52, in add_component
    device.add_component(component_config)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 94, in run_device_legacy
    run_device_legacy(create_legacy_device_config(power_path), component_config)
  File "/var/www/html/openWB/packages/modules/http/device.py", line 117, in read_legacy_bat
    parser.set_defaults(RUN=lambda args: function(*[getattr(args, argument_name) for argument_name in arg_spec.args]))
  File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 19, in <lambda>
    args.RUN(args)
  File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 33, in run_using_positional_cli_args
    {"bat": read_legacy_bat, "counter": read_legacy_counter, "inverter": read_legacy_inverter}
  File "/var/www/html/openWB/packages/modules/http/device.py", line 146, in <module>`

yankee42, 2021-12-20:
Tipp am Rande: Wenn du drei Backticks nimmst (```), statt nur einen, dann klappt auch das Formatieren von solchen Blöcken.

matzempc, 2021-12-20:
> Habe den PR #1844 gerade mal manuell getestet @matzempc . Also Netz und Haus sieht wieder gut aus aber beim Speicher gibt es noch KeyError:

Sollte jetzt auch gefixt sein (sowie auch gleich das inverter modul)! Habe es mit in den PR #1844 mit aufgenommen!

xerion3800, 2021-12-20:
Sieht gut aus habe es gerade mal kurz für den Speicher testen können.

matzempc, 2021-12-21:
So dann sollte alles behoben sein, mein PR wurde vorhin gemergt und EVU und PV funktioniert bei mir, da du den Speicher (bat) testen konntest passt alles wieder.
Das Problem mit "none" als URL wurde von @yankee42 auch gelöst => geht bei mir jetzt wieder ohne "fake"-URL!

closed 2021-12-21 by xerion3800