snaptec/openWB #1848: Neues/angepasstes Modul für SolaX 3phasen Hybrid WR

AndreasRauhut, 2021-12-21:
Im Forum wurde herausgearbeitet, dass das vorhandene SolaX WR Modul nur für 1-phasige WR funktioniert. 
Gern würde ich auch meinen 3-phasigen Hybrid WR an openWB anbinden. 
Die notwendigen Anpassungen hat Peete hier formuliert. [https://openwb.de/forum/viewtopic.php?p=52181#p52181](https://openwb.de/forum/viewtopic.php?p=52181#p52181)

benderl, 2021-12-25:
Am Besten einen PR mit den Anpassungen stellen.

AndreasRauhut, 2022-01-07:
Bitte wo kann ich nachlesen, wie ein PR( ich denke das heißt: Pull Request) gestellt wird. Im openWB Forum wird auch darüber geschrieben, dass ein solcher gestellt wird. Wer kann das machen und welche Informationen bedarf es ?

benderl, 2022-02-08:
Das Thema hat sich inzwischen geklärt, der WR wird unterstützt.

AndreasRauhut, 2022-02-09:
Es besteht noch ein Anpassungsbedarf bezüglich des Werts des Gesamtertrags WR1.
Hier muß der in openWB gezeigte Wert noch ,mit 1000 multipliziert werden
Siehe [https://openwb.de/forum/viewtopic.php?p=57672#p57672](https://openwb.de/forum/viewtopic.php?p=57672#p57672)

AndreasRauhut, 2022-02-11:
Ein weiterer Punkt ist die Herkunft der auf der Status-Seite in EVU angezeigten Spannungswerte für Phase 1,2,3.
Diese scheinen im Code als Konstanten mit 230V eingetragen zu sein. Ich kann nicht nachvollziehen, ob diese Werte auch zur weiteren Berechnung genutzt werden.
Vielleicht kann man diese durch die im modbus des WR gelesenen Werte ersetzen.

benderl, 2022-02-25:
Der WR liefert lediglich eine Spannung für alle drei Phasen. Die Info bietet also keinen Mehrwert und wird von openWB intern nicht weiter benötigt.

closed 2022-02-25 by benderl

AndreasRauhut, 2022-02-25:
Danke